SatDump/SatDump #620: Tracker fails with log "(C) AOS!!!!!!!!!!!!!! 33591"

truglodite, 2024-02-01:
**Description of the issue**
Tracking schedule fails when a tracked satellite starts a pass. The log shows "<timestamp> (C) AOS!!!!!!!!!!!!!! 33591" when this happens, and processing for the pass does not start.

**Hardware (SDR/PC/OS)**
Pi4b w/ Rasbian bookworm 64.

**Version (Eg, 1.0.0, CI Build 171)**
I installed the precompiled satdump version 1.1.3.

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)
```
[15:58:50 - 01/02/2024] (T) Update pass trajectory...
[17:26:19 - 01/02/2024] (C) AOS!!!!!!!!!!!!!! 33591
[17:26:19 - 01/02/2024] (T) Updating elements...
[17:26:19 - 01/02/2024] (T) Update pass trajectory...
```

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**
TLE's appear to be updating properly, indicated by message in the bottom of UI and

truglodite, 2024-02-02:
Wow I feel humiliated… I simply forgot to check the “Live” boxes for the tracked satellites.

closed 2024-02-02 by truglodite

Aang23, 2024-02-04:
No worries... Happens to everyone.

The "Error" message was made this way just so it would trigger something obvious while debugging - as this is still WIP.