SatDump/SatDump #621: Cmake error Ubuntuq

f6bvp, 2024-02-05:
In README.md

### Linux

On Linux, building from source is recommended, but builds are provided for x64-based Ubuntu distributions. Here are some generic (Debian-oriented) build instructions.

```bash
# Install dependencies on Debian-based systems:
#sudo apt install git build-essential cmake g++ pkgconf libfftw3-dev libvolk2-dev libpng-dev libjemalloc-dev   # Core dependencies. If libvolk2-dev is not available, use libvolk1-dev

Since a few days I get errors when performing cmake

JVital2013, 2024-02-06:
Have you switched Ubuntu versions recently? The ubuntu package for volk on all versions of Ubuntu up to 23.04 must have the 1 or 2. It appears it has changed for Ubuntu 23.10 to simply be libvolk-dev

JVital2013, 2024-02-07:
I installed Ubuntu 23.10 and confirmed the package change. It looks like it's going to be a mixed bag for a bit - Ubuntu has moved to `libvolk-dev`, but Debian Bookworm is still `libvolk2-dev`. It appears they are planning on moving to `libvolk-dev` as well, though, in the future.

Since most systems would still be running Debian/Ubuntu versions with `libvolk2-dev`, I updated the readme to list `libvolk2-dev` first, followed by `libvolk-dev` (only the newest Ubuntu right now), and finally `libvolk1-dev` (for old systems). Once libvolk-dev becomes more prevalent, we can re-order the README again.

closed 2024-02-07 by JVital2013

f6bvp, 2024-05-23:
Just installed Ubuntu 24-04 LTS
https://github.com/SatDump/SatDump#linux
**Still displays  :** 

# Install dependencies on Debian-based systems:
sudo apt install git build-essential cmake g++ pkgconf libfftw3-dev libpng-dev libtiff-dev libjemalloc-dev   # Core dependencies
sudo apt install libvolk2-dev  

However, as already reported libvolk2-dev does not exist on Ubuntu and libvolk-dev must be installed.
README should be really updated.

After that make is ok.