SatDump/SatDump #624: NOAA 15 HRPT amsu_a crash

krakenrf, 2024-02-06:
Getting this crash at least once or twice a day when trying to run Satdump with the scheduler. I believe it's a problem with an HRPT product from a NOAA 15. There are a bunch of maybe unrelated GPU errors in the log, before the whole program closes with an Abort.

Running an Orange Pi 5 with Ubuntu 22.04.

If it's helpful I can try and record NOAA 15 next time.

```
[07:15:04 - 06/02/2024] (D) Composite needs calibrated channel ch1
[07:15:04 - 06/02/2024] (T) Cached calibrated image channel 1
[07:15:04 - 06/02/2024] (D) Composite needs calibrated channel ch3a
[07:15:04 - 06/02/2024] (T) Cached calibrated image channel 3
[07:15:04 - 06/02/2024] (D) Composite needs calibrated channel ch4
[07:15:04 - 06/02/2024] (T) Cached calibrated image channel 5
[07:15:04 - 06/02/2024] (T) Max Total Width 2048
[07:15:04 - 06/02/2024] (T) Max Total Used  2048
[07:15:04 - 06/02/2024] (T) Offset for cch1 is 0
[07:15:04 - 06/02/2024] (T) Offset for cch3a is 0
[07:15:04 - 06/02/2024] (T) Offset for cch4 is 0
[07:15:04 - 06/02/2024] (I) Saving /media/orangepi/WX Images/live_output/2024-02-06_07-00_noaa_hrpt_1.7025 GHz/AVHRR/avhrr_3_rgb_Day_Microphysics.png...
[07:15:04 - 06/02/2024] (I) Saving /media/orangepi/WX Images/live_output/2024-02-06_07-00_noaa_hrpt_1.7025 GHz/AVHRR/avhrr_3_rgb_Day_Microphysics_corrected.png...
[07:15:04 - 06/02/2024] (D) Composite needs calibrated channel ch4
[07:15:04 - 06/02/2024] (T) Cached calibrated image channel 5
[07:15:04 - 06/02/2024] (T) Max Total Width 2048
[07:15:04 - 06/02/2024] (T) Max Total Used  2048
[07:15:04 - 06/02/2024] (T) Offset for cch4 is 0
[07:15:04 - 06/02/2024] (D) Filtering timestamps...
[07:15:07 - 06/02/2024] (I) Saving /media/orangepi/WX Images/live_output/2024-02-06_07-00_noaa_hrpt_1.7025 GHz/AVHRR/avhrr_3_rgb_MCIR.png...
[07:15:07 - 06/02/2024] (D) Filtering timestamps...
[07:15:07 - 06/02/2024] (I) Solving TPS equations...
[07:15:07 - 06/02/2024] (I) Forward solved
[07:15:07 - 06/02/2024] (E) Error generating transformer!
[07:15:07 - 06/02/2024] (I) Drawing map overlay...
[07:15:07 - 06/02/2024] (I) Saving /media/orangepi/WX Images/live_output/2024-02-06_07-00_noaa_hrpt_1.7025 GHz/AVHRR/avhrr_3_rgb_MCIR_map.png...
[07:15:07 - 06/02/2024] (D) Composite needs calibrated channel ch1
[07:15:07 - 06/02/2024] (T) Cached calibrated image channel 1
[07:15:07 - 06/02/2024] (T) Max Total Width 2048
[07:15:07 - 06/02/2024] (T) Max Total Used  2048
[07:15:07 - 06/02/2024] (T) Offset for cch1 is 0
[07:15:07 - 06/02/2024] (D) Filtering timestamps...
[07:15:09 - 06/02/2024] (I) Saving /media/orangepi/WX Images/live_output/2024-02-06_07-00_noaa_hrpt_1.7025 GHz/AVHRR/avhrr_3_rgb_MSA_(channel_1).png...
[07:15:09 - 06/02/2024] (I) Saving /media/orangepi/WX Images/live_output/2024-02-06_07-00_noaa_hrpt_1.7025 GHz/AVHRR/avhrr_3_rgb_MSA_(channel_1)_corrected.png...
[07:15:09 - 06/02/2024] (I) Drawing map overlay...
[07:15:09 - 06/02/2024] (I) Saving /media/orangepi/WX Images/live_output/2024-02-06_07-00_noaa_hrpt_1.7025 GHz/AVHRR/avhrr_3_rgb_MSA_(channel_1)_map.png...
[07:15:09 - 06/02/2024] (I) Drawing map overlay...
[07:15:09 - 06/02/2024] (I) Saving /media/orangepi/WX Images/live_output/2024-02-06_07-00_noaa_hrpt_1.7025 GHz/AVHRR/avhrr_3_rgb_MSA_(channel_1)_corrected_map.png...
[07:15:09 - 06/02/2024] (D) Composite needs calibrated channel ch2
[07:15:09 - 06/02/2024] (T) Cached calibrated image channel 2
[07:15:09 - 06/02/2024] (T) Max Total Width 2048
[07:15:09 - 06/02/2024] (T) Max Total Used  2048
[07:15:09 - 06/02/2024] (T) Offset for cch2 is 0
[07:15:09 - 06/02/2024] (D) Filtering timestamps...
[07:15:12 - 06/02/2024] (I) Saving /media/orangepi/WX Images/live_output/2024-02-06_07-00_noaa_hrpt_1.7025 GHz/AVHRR/avhrr_3_rgb_MSA_(channel_2).png...
[07:15:12 - 06/02/2024] (I) Saving /media/orangepi/WX Images/live_output/2024-02-06_07-00_noaa_hrpt_1.7025 GHz/AVHRR/avhrr_3_rgb_MSA_(channel_2)_corrected.png...
[07:15:12 - 06/02/2024] (I) Drawing map overlay...
[07:15:12 - 06/02/2024] (I) Saving /media/orangepi/WX Images/live_output/2024-02-06_07-00_noaa_hrpt_1.7025 GHz/AVHRR/avhrr_3_rgb_MSA_(channel_2)_map.png...
[07:15:12 - 06/02/2024] (I) Drawing map overlay...
[07:15:12 - 06/02/2024] (I) Saving /media/orangepi/WX Images/live_output/2024-02-06_07-00_noaa_hrpt_1.7025 GHz/AVHRR/avhrr_3_rgb_MSA_(channel_2)_corrected_map.png...
[07:15:12 - 06/02/2024] (D) Composite needs calibrated channel ch4
[07:15:12 - 06/02/2024] (T) Generating calibrated image channel 5. Range 4.261273 155.278282. Type 2
[07:15:14 - 06/02/2024] (T) Max Total Width 2048
[07:15:14 - 06/02/2024] (T) Max Total Used  2048
[07:15:14 - 06/02/2024] (T) Offset for cch4 is 0
[07:15:14 - 06/02/2024] (I) Saving /media/orangepi/WX Images/live_output/2024-02-06_07-00_noaa_hrpt_1.7025 GHz/AVHRR/avhrr_3_rgb_NO_enhancement.png...
[07:15:14 - 06/02/2024] (I) Drawing map overlay...
[07:15:14 - 06/02/2024] (I) Saving /media/orangepi/WX Images/live_output/2024-02-06_07-00_noaa_hrpt_1.7025 GHz/AVHRR/avhrr_3_rgb_NO_enhancement_map.png...
[07:15:14 - 06/02/2024] (D) Skipping APT channel A as it can't be made!
[07:15:14 - 06/02/2024] (D) Skipping APT channel B as it can't be made!
[07:15:14 - 06/02/2024] (D) Skipping MCIR (Uncalibrated) as it can't be made!
[07:15:14 - 06/02/2024] (D) Skipping MSA (Uncalibrated) as it can't be made!
[07:15:14 - 06/02/2024] (D) Skipping MCIR Rain (Uncalibrated) as it can't be made!
[... 330 lines cut ...]
[07:15:17 - 06/02/2024] (I) Found OpenCL Devices (1) :
[07:15:17 - 06/02/2024] (D)  - Mali-LODX r0p0
[07:15:17 - 06/02/2024] (I) /media/orangepi/WX Images/live_output/2024-02-06_07-00_noaa_hrpt_1.7025 GHz/AMSU/product.cbor
[07:15:17 - 06/02/2024] (I) /media/orangepi/WX Images/live_output/2024-02-06_07-00_noaa_hrpt_1.7025 GHz/AMSU/product.cbor
[07:15:17 - 06/02/2024] (I) Loading AMSU-A-1.png
[07:15:17 - 06/02/2024] (I) Loading AMSU-A-2.png
[07:15:17 - 06/02/2024] (I) Loading AMSU-A-3.png
[07:15:17 - 06/02/2024] (I) Loading AMSU-A-4.png
[07:15:17 - 06/02/2024] (I) Loading AMSU-A-5.png
[07:15:17 - 06/02/2024] (I) Loading AMSU-A-6.png
[07:15:17 - 06/02/2024] (I) Loading AMSU-A-7.png
[07:15:17 - 06/02/2024] (I) Loading AMSU-A-8.png
[07:15:17 - 06/02/2024] (I) Loading AMSU-A-9.png
[07:15:17 - 06/02/2024] (I) Loading AMSU-A-10.png
[07:15:17 - 06/02/2024] (I) Loading AMSU-A-11.png
[07:15:17 - 06/02/2024] (I) Loading AMSU-A-12.png
[07:15:17 - 06/02/2024] (I) Loading AMSU-A-13.png
[07:15:17 - 06/02/2024] (I) Update image...
[07:15:17 - 06/02/2024] (I) Done
[07:15:17 - 06/02/2024] (I) Loading AMSU-A-14.png
[07:15:17 - 06/02/2024] (I) Loading AMSU-A-15.png
[07:15:17 - 06/02/2024] (D) Using handler image_handler for instrument amsu_a
satdump-ui: /home/orangepi/SatDump/plugins/../src-core/nlohmann/json.hpp:21357: const value_type& nlohmann::json_abi_v3_11_2::basic_json<ObjectType, ArrayType, StringType, BooleanType, NumberIntegerType, NumberUnsignedType, NumberFloatType, AllocatorType, JSONSerializer, BinaryType, CustomBaseClass>::operator[](const typename nlohmann::json_abi_v3_11_2::basic_json<ObjectType, ArrayType, StringType, BooleanType, NumberIntegerType, NumberUnsignedType, NumberFloatType, AllocatorType, JSONSerializer, BinaryType, CustomBaseClass>::object_t::key_type&) const [with ObjectType = std::map; ArrayType = std::vector; StringType = std::__cxx11::basic_string<char>; BooleanType = bool; NumberIntegerType = long int; NumberUnsignedType = long unsigned int; NumberFloatType = double; AllocatorType = std::allocator; JSONSerializer = nlohmann::json_abi_v3_11_2::adl_serializer; BinaryType = std::vector<unsigned char>; CustomBaseClass = void; nlohmann::json_abi_v3_11_2::basic_json<ObjectType, ArrayType, StringType, BooleanType, NumberIntegerType, NumberUnsignedType, NumberFloatType, AllocatorType, JSONSerializer, BinaryType, CustomBaseClass>::const_reference = const nlohmann::json_abi_v3_11_2::basic_json<>&; typename nlohmann::json_abi_v3_11_2::basic_json<ObjectType, ArrayType, StringType, BooleanType, NumberIntegerType, NumberUnsignedType, NumberFloatType, AllocatorType, JSONSerializer, BinaryType, CustomBaseClass>::object_t::key_type = std::__cxx11::basic_string<char>; nlohmann::json_abi_v3_11_2::basic_json<ObjectType, ArrayType, StringType, BooleanType, NumberIntegerType, NumberUnsignedType, NumberFloatType, AllocatorType, JSONSerializer, BinaryType, CustomBaseClass>::object_t = std::map<std::__cxx11::basic_string<char>, nlohmann::json_abi_v3_11_2::basic_json<>, std::less<void>, std::allocator<std::pair<const std::__cxx11::basic_string<char>, nlohmann::json_abi_v3_11_2::basic_json<> > > >]: Assertion `it != m_value.object->end()' failed.
Aborted (core dumped)
```

Aang23, 2024-02-06:
I'll admit this isn't too unlikely - I'll do some testing.

JVital2013, 2024-02-08:
Hi @krakenrf, this should be fixed now. Can you update to the latest nightly and try again?

krakenrf, 2024-02-10:
@JVital2013 Thanks, I'll run it for a few days to confirm.

krakenrf, 2024-02-13:
Appears to be fixed, closing. Thanks!

closed 2024-02-13 by krakenrf