SatDump/SatDump #625: satdump-ui: symbol lookup error: satdump-ui: undefined symbol:

f6bvp, 2024-02-06:
I performed a complete environment install on Ubuntu according to README file.

Download of daily build is ok.
All compilation is ok.

satdump_sdr_server runs correctly

However when starting satdump-ui fails with error message :+1: 

satdump-ui: symbol lookup error: satdump-ui: undefined symbol: _ZN7backend12device_scaleE

grep -r ZN7backend12device

only finds occurence in binary code...

f6bvp, 2024-02-06:
Compiling project with debug option and ggdb2 here is the error location :+1: 
Reading symbols from /usr/bin/satdump-ui...
(gdb) run
Starting program: /usr/bin/satdump-ui 
/usr/bin/satdump-ui: symbol lookup error: /usr/bin/satdump-ui: undefined symbol: _ZN7backend12rebuildFontsE
[Inferior 1 (process 41938) exited with code 0177]
(gdb) where
No stack.
(gdb) list
26	{
27	    logger->error("Glfw Error " + std::to_string(error) + ": " + description);
28	}
29	
30	void window_content_scale_callback(GLFWwindow *, float xscale, float)
31	{
32	    backend::device_scale = xscale / style::macos_framebuffer_scale();
33	    satdump::update_ui = true;
34	}
35	
(gdb)

JVital2013, 2024-02-06:
You can try deleting the contents of your build folder and running cmake from scratch, but you likely already did that if you built it as debug.

To confirm, based on what you said in your other issue: have you switched to Ubuntu 23.10? Any special gcc versions? We may need to spin up a VM and test this version

JVital2013, 2024-02-06:
I ran a quick test and Ubuntu 23.10 seems to work fine, other than the dependency changes you mention in your other issue. I look forward to hearing if clearing the build folder resolves the issue, or if you are using a custom gcc of some type.

[image]

f6bvp, 2024-02-08:
On my remote Ubuntu system satdump install is ok with following settings : 

Linux version 6.5.3 (root@gpredict) (gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #3 SMP PREEMPT_DYNAMIC Mon Feb  5 20:38:44 CET 2024

gcc version 11.4.0 (Ubuntu 11.4.0-1ubuntu1~22.04) 

libvolk2-dev/jammy,now 2.5.1-1 amd64  [installé]

I cannot run satdump-ui for 
 (E) Glfw Error 65544: X11: The DISPLAY environment variable is missing
[11:16:31 - 08/02/2024] (C) Could not init GLFW

despite accessing with ssh -Y

However this is not important.

f6bvp, 2024-02-08:
On my local system satdump-ui is working fine.

Linux version 6.8.0-rc2-DEBUG+ (root@bernard-f6bvp) (gcc (Ubuntu 13.2.0-4ubuntu3) 13.2.0, GNU ld (GNU Binutils for Ubuntu) 2.41) #10 SMP PREEMPT_DYNAMIC Mon Feb  5 12:38:05 CET 2024

gcc version 13.2.0 (Ubuntu 13.2.0-4ubuntu3) 

ibvolk-dev/mantic,now 3.0.0-2 amd64  [installé]

Thanks for your help.

NB : kernel versions are different for I am compiling Linux kernels for hamradio network modules debugging and testing purpose (AX.25, ROSE, etc..). This is my main hobby.

f6bvp, 2024-02-08:
Here is the configuration of the third system I am working with :  

Linux version 5.15.0-92-generic (buildd@lcy02-amd64-002) (gcc (Ubuntu 9.4.0-1ubuntu1~20.04.2) 9.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #102~20.04.1-Ubuntu SMP Mon Jan 15 13:09:14 UTC 2024

gcc version 9.4.0 (Ubuntu 9.4.0-1ubuntu1~20.04.2) 

apt list -a libvolk2-dev
En train de lister... Fait
libvolk2-dev/focal,now 2.4.1-2~ubuntu20.04.1~ppa1 amd64  [installé]
libvolk2-dev/focal 2.2.1-1 amd64

I am also logging via ssh -Y and when starting satdump-ui I got the previously reported error : 

satdump-ui: symbol lookup error: satdump-ui: undefined symbol: _ZN7backend12device_scaleE

Then I removed  and reinstalled libvolk2-dev for there are no libvolk1-dev nor libvolk-dev available : 

Réception de :1 http://ppa.launchpad.net/gnuradio/gnuradio-releases-3.9/ubuntu focal/main amd64 libvolk2-dev amd64 2.4.1-2~ubuntu20.04.1~ppa1 [393 kB]
393 ko réceptionnés en 0s (1 424 ko/s) 
Sélection du paquet libvolk2-dev:amd64 précédemment désélectionné.
(Lecture de la base de données... 304604 fichiers et répertoires déjà installés.)
Préparation du dépaquetage de .../libvolk2-dev_2.4.1-2~ubuntu20.04.1~ppa1_amd64.deb ...
Dépaquetage de libvolk2-dev:amd64 (2.4.1-2~ubuntu20.04.1~ppa1) ...
Paramétrage de libvolk2-dev:amd64 (2.4.1-2~ubuntu20.04.1~ppa1) ...

 satdump-ui still displays the same error : 
satdump-ui: symbol lookup error: satdump-ui: undefined symbol: _ZN7backend12device_scaleE

JVital2013, 2024-02-08:
Thanks for the additional info!

The symbols its complaining about should be in libsatdump_core.so. I'm suspicious it's loading an old copy of libsatdump_core.so from somewhere, instead of the recently-build one. These symbols have only recently been added.

Please make sure there are no stray copies of libsatdump_core.so in any of the following places:

- In the same folder as satdump_ui
- In the current working direcory
- /usr/lib
- /lib
- Whatever you set CMAKE_INSTALL_PREFIX to (`CMAKE_INSTALL_PREFIX/lib`)

For your reference: `_ZN7backend12rebuildFontsE -> backend::rebuildFonts()` and `_ZN7backend12device_scaleE -> backend::device_scale`

f6bvp, 2024-02-12:
I deleted all occurence of libsatdump_core.so
and rebuild satdump daily source.

find / -name libsatdump_core.so
did not find any.

And after rebuilding satdump on my different machines everything is ok.

Many thanks !
I hope I did not took much of your time.

Regards,
Bernard

closed 2024-02-12 by f6bvp