SatDump/SatDump #628: MCIR Rain (Uncalibrated) based on dataset.json with CLI

iHD992, 2024-02-10:
Is it possible to generate a composite with the CLI based on the live generated data?

Example:
I have: APT-A.png, APT-B.png, AVRR-2.png, AVRR-4.png, dataset.json, product.cbor and raw.png
I need: MCIR Rain (Uncalibrated) 
Bonus: Stereo-Projection

I could not find something similar in the docs.

iHD992, 2024-02-16:
Starting from baseband recording, this is not a problem:
.\satdump noaa_apt baseband [input_file] [output_folder] --satellite_number [number]

Is there something like ".\satdump noaa_apt livedata product.cbor [output_folder]"?

The baseband files are quite huge, would be easier to just store ~20 MB of live data.

RAD750, 2024-02-18:
I somehow missed this, sorry :(
You can use the user settings.json file to specify what composites to autogenerate. If you want only MCIR Rain, just copy every other AVHRR/3 composite from the main config.json and set `autogen: false` for all of them, except the ones you want! At the same time, you can add in the stereo projection options as detailed in the doc.

Here's the relevant documentation for both of your questions:
https://docs.satdump.org/composites.html#overriding-the-default-configuration

It's not the most easy to explain, so if you need help or you have suggestions on how to improve the documentation, I'd be more than happy to hear them or help you out!

iHD992, 2024-02-18:
The autogen feature is nice, but only helps me when I know in advance what I want.

Maybe I have to elaborate further with an example:
I have a Raspberry Pi gathering the data from NOAA APT automatically for something like video over a long time (say one picture a day). There are several problems;
1. As the device is not that powerful and short on storage I would prefer not to save the baseband data but the smallest live data possible. 
2. As I am not sure about the "viewing angle" (projection) or the best composite, being flexible would also be nice. 
3. As I have a lot of passes in the end, using the GUI is tedious. Thus using the CLI (and scripting its usage with a python script) is the better way.

When I capture live data of NOAA DSB I get a ".tip" and if I capture live data of a METEOR I get a ".cadu". The first can later be processed with the noaa_dsb (frames) or the meteor_m2_lrpt (cadu) pipelines using the CLI (not tried, but pretty sure it works). Thus I can capture a lot of small ".tip" and ".cadu" and generate "perfect" composites and projections a year later using the CLI.

When I capture live data of NOAA APT, I get APT-A.png, APT-B.png, AVRR-X.png, AVRR-Y.png, dataset.json, product.cbor and raw.png. It looks to me like I can not feed these files into the CLI, thus in case of NOAA APT only the GUI is possible using the viewer tab. Am I right?

If my problem is still not clear. I could draw some figures to explain, if necessary.

About the documentation. Here some problems I ran into:
1. https://docs.satdump.org/composites.html#projections : Stereo needs center_lat, center_lon and scale not lat, lon and scale.
2. https://docs.satdump.org/composites.html#overriding-the-default-configuration: I could not get the overwrite working using settings.json based on the docu. At the moment I stick with modifying satdump_cfg.json.
3. https://docs.satdump.org/composites.html#advanced: "If omitted, they will be treated as false (for example, if you don’t specify "equalize": true, the image will not be equalized). Forcefully setting these parameters to false is redundant." this is not true for autogen. The default for autogen is "true".
4. I am using the autotrack.json. The best description I found here: https://www.a-centauri.com/articoli/automating-satellite-reception-with-satdump-cli . Would be nice to have something similar in the docu. Really important for me were             "autotrack_min_elevation": 20 and "multi_mode": true. Both are not described in the docu or on www.a-centauri.com/articoli/.