SatDump/SatDump #631: Radio Astronomy Support

krakenrf, 2024-02-14:
This might be only loosely related to satellites, but I just wanted to know if anyone working on Satdump has considered adding support for radio astronomy, specifically hydrogen line observations? It seems like Satdump would be a good candidate for an easy to use RA program, as it already has the rotor tracking, FFT and various SDR support required.

What I would love to see is:

1. The ability to average the FFT for any amount of time while tracking a point in the sky
2. The ability to record the spectrum over time, or at different rotor positions, and record the power level and the frequency of the max power to see the doppler shifts at different parts of the milky way.
3. The ability to know the current sky map for the users location
4. And the ability to move a rotator to a specific position in various astronomical coordinate systems, and scan the entire sky or galactic plane (or parts of) over time via a tracking mount.

I'm sure more advanced features like the ability to build a pixel map of observations to create a H-line galactic image, and ways to incorporate known dish parameters like beamwidth could be added too.

Aang23, 2024-02-15:
Hello! To be honest it is not that loosely related, and most of what you mentioned does make sense for satellite tracking to some extent. Making it possible to do long acquisitions and average a FFT properly etc in order to see weak signals is also useful for DSN, finding weak downlinks, etc.

Hence, your suggestions for 1, 2, 3 as well as better controls to do very high levels of FFT averaging 100% make sense to implement. 4 and 5 make sense at the very least by being able to show / input coordinates in Ra/Dec, as this is also often used for DSN and other similar targets. That does more or less align with what I was intending on doing anyway.

I can't give an ETA but those will be coming at some point. Though more specific features will need some more thinking.

berniskijump, 2025-01-31:
Hello, I stumbled over this thread and due to my interest lying in both RA and sat tracking I did some thinking and want to share my thoughts and ideas. I devised some more specific feature description that may help.

1. if not exists a logging Pipline in which the raw RF signal is  "annotated" by the antenna position(AzElLog)
2. if not exists RF signal processor pipelines (maybe as plugins) which turns raw RF signals into number streams, This can include but is not limited to signal power, SNR, doppler shift, etc.
3. if not exists earth rotation correction algorithm plugin 
4. Open-loop dish pointing pattern, think sky scanning with no RF feedback. sky rotation correction  can be applied, useful for Trouble shooting dish site, finding GEO Sats and RA
5. Closed-loop dish Pointing pattern, think sat finding pattern based on RF feedback to estimate orbit of unknown sats or hone in on GEO sats
6. data rendering for site sky,  Colored Polar plot would be my personal preference
7. data rendering of galactical horizon, similar to sat image viewer and projection

in terms of location I believe it is best when: 
1 is under the recorder tab as a checkbox(AzElLog), 2 is present as a select-able under recorder => tracker => bullet point "general"  in case of closed loop dish guiding selected and for the data rendering interfaces(6,7), 3 shall be a configuration point of feature 4 and 7, 4 and 5 under recorder => tracker => bullet point "general" as a select-able with mentioned and assumed configs, 6 and 7 under Viewer => Observations

Proposed workflow for couple of scenarios:
RA; go to recorder-> configure as required (freq, device, etc.), select AzElLog, select open loop scanning with earth rotation compensation activated, select mapping procedure -> start recording -> When finished go to observation, select recording(s)-> configure signal processor-> compute and display with galactic processor
Spacial Reception quality check; go to recorder -> configure as required, select logging-> record -> many passes recorded, go to observation -> select recordings, select processor(e.g. SNR) -> compute and display in polar plot view
Sat finder; go to recorder -> configure as required, select AzElLog, select closed loop scanning, select processor for adaptive search, select search pattern, select expected entry point -> arm/activate -> hope it locks on  something -> if lock achived, go to Observations -> select recording, select processor -> print and analyse in polar view
dish site noise finder: go to recorder-> configure as required (freq, device, etc.), select AzElLog, select open loop scanning,  select mapping procedure -> start recording -> When finished go to observation, select recording-> configure signal processor-> compute and display in polar plot

My final thoughts. I might have inadvertently added some additional features not strictly within the first proposed scope. But due to my overlapping interests it snowballed a bit. In my view I mostly addressed  points 2-5 by @krakenrf . The first point was something my brain couldn't really wrap itself around. I thought about pre-processors but could not find a good place to fit them. I just want to add that I most likely missed some details. Additionally I am more than happy for comments and critique. Also, my naming scheme  might not be perfect. Hope my couple of brain cycles printed to text  help someone.