SatDump/SatDump #633: Crash when trying to decode NOAA-19 from WAV file

rubberduck203, 2024-02-17:
We chatted a bit on Reddit. Apologies it's taken me a few days to sit down and write this up.

**Description of the issue** 

I recorded an *.m4a file of NOAA-19 and converted it to WAV using Audacity with an 11025Hz sample rate. 
`NOAA-19.WAV` in the attached *.zip file.  
It is very low quality, but I was able to decode this image using https://apt.open-weather.community/ 

[image]

The settings I used below, I made sure to uncheck "auto" for the timestamp because I saw other issues related to using it.  
The crash occurs whether or not it is checked.

[image]

When I click start, several files are generated, `dataset.json`, `product.cbor`,  `raw.png`, `APT-A.png`, and `APT-B.png`.  
These are also in the attached archive.   

Inspecting the logs indicates that SatDump crashes immediately after generating `APT-B.png`.  
(Full log file are also in attached archive.)

```console
...
[10:44:45 - 17/02/2024] (C) Couldn't calibrate image with wedges! Likely no valid wedge was identified!
[10:44:45 - 17/02/2024] (I) Saving /Users/rubberduck/Documents/noaa/raw.png...
[10:44:45 - 17/02/2024] (I) ----------- Satellite
[10:44:45 - 17/02/2024] (I) NORAD : 33591
[10:44:45 - 17/02/2024] (I) Name  : NOAA-19
[10:44:45 - 17/02/2024] (I) ----------- AVHRR
[10:44:45 - 17/02/2024] (I) Lines : 343
[10:44:45 - 17/02/2024] (W) Identified channels (debug): A: -1 A1: -1 B: -1
[10:44:45 - 17/02/2024] (I) Saving /Users/rubberduck/Documents/noaa/APT-A.png...
[10:44:45 - 17/02/2024] (I) Saving /Users/rubberduck/Documents/noaa/APT-B.png...
```

**Hardware (SDR/PC/OS)**

Mac Big Sur 11.7.9 

**Version (Eg, 1.0.0, CI Build 171)**

v1.1.3
I tried the nightly build, but that was immediately crashing when I tried to start it.

** All the relevant files I could dig up **

[noaa.zip](https://github.com/SatDump/SatDump/files/14318557/noaa.zip)

JVital2013, 2024-02-17:
If you switch to the latest nightly, it shouldn't crash anymore - this has been fixed.

That being said, I see from your [Reddit post](https://www.reddit.com/r/amateursatellites/comments/1apr3e1/noaa19_it_can_only_get_better_from_here/) and [the other discussion](https://www.reddit.com/r/amateursatellites/comments/1aogqjz/i_heard_noaa19/) with @RAD750. The problem here is that you recorded this with a HT and your phone's microphone.

I'll be the first to admit, getting satellites like this can feel magical, especially the first few times. However - and I'll let others in this project chime in here - this will not be supported. It's not possible to get enough bandwidth like this  in the first place - never mind other issues, such as wind noise, audio compression artifacts from the losst m4a format, etc. I'd recommend getting an SDR - some of which cost less than decent HTs - and would result in much cleaner images.

[image]

rubberduck203, 2024-02-17:
Yeah @JVital2013 that’s fine and all. I don’t expect to get a _good_ image that way, but FWIW it is going to be how 99% of people get their first one, so you should consider that as well as the fact that other tools _can_ decode the image. The latter alone points to something being not quite right. I was mostly just trying to give you a good report of the crash I was experiencing. I noticed in other issues an attitude of “oh, you used the wrong pipeline and it crashed”, which is a pretty awful UX. If you really want to build something to replace several other pieces of software, I would suggest a more defensive style of programming. 

I would also recommend you be nicer. I was considering actually contributing, but you’re not very nice and I don’t really want to spend my free time working with you. 

Good luck.   
Blessed be.

Aang23, 2024-02-17:
Just to chime in. The reason @JVital2013 is saying using a HT this way is "not supported" is for some technical reasons :
1) Most HT meant for amateur radio communications will only do NFM up to ~25-30k of bandwidth. That'll let you hear the APT signal to some extent, but the minimum and maximum values will be truncated rendering a lot of what's necessary to properly process the data useless. APT uses ~40k of bandwidth including the minimal effects of Doppler shift. This could be fixed by modifying your HT's hardware of firmware (or using WFM if you can).

2) Recording via a microphone will further distort the signal, as both the speaker's frequency response, the air (and additional noise such as wind, you holding the HT, etc), and microphone's response will further distort it. One way to avoid this would be to hook up your HT's output via an audio cable instead.

3) When the data is distorted like this, the "lines" you see in the image, used for synchronizing and processing the image won't be usable. Some software will fail to process and not synchronize at all - such as the website you linked, but this makes the data worthless as far as using it in any proper way goes (and nowadays, pretty much everyone uses proper SDR options from the start). Though it is a very good point it allows seeing an image "at all" - so a workaround we could implement is an option to save the un-synced image anyway.

Otherwise, while I can see how you may find SatDump a bit less obvious to use for APT than some other options, we also support a very wide range of satellite signals just as much. It's not obvious how this could be much easier than it is now... But I'm open to any ideas! :-)

But for all other issues - apart from adding an option to dump the un-synced image anyway - you will not get proper processing without upgrading the hardware you are using, with a RTL-SDR or similar as it's been mentioned already for example. At this point, with a proper reception chain you'll also be able to see more of the processing features we've implemented to provide an alternative to previous options, such as calibration, projections, and more.

rubberduck203, 2024-02-18:
I want to be 100% clear about this…

I have zero interest in arguing with anyone over the scope of this project. I said I would provide a bug report for the _crash_ and that’s all I came to do. 

I’ll give the latest nightly a shot when I have a moment.

JVital2013, 2024-02-18:
Hello again,

After some testing, we have added saving unsynced images by default as well. When SatDump can't synchronize the image for whatever reason, this gives a much better result - hopefully you agree! Here is your image with the new changes - both A and B channels. It will be saved as raw_unsync:

[image]

JVital2013, 2024-03-23:
Closing this issue as the original crash was fixed.

An issue like this has thankfully not come up before, so I'm going to make an example out of it in case it comes up again. **For future visitors here: behavior like this is not acceptable.**

## Things that happened in the issue that were acceptable:
- Using an HT to record the baseband (making newbie decisions are acceptable, of course!)
- Asking for help in using the program
- Reporting a crash
- Going radio silent, and not following up (we all get busy, or move on!)

## What was not acceptable

### The inability to accept constructive criticism
In this issue, we were giving feedback on what works better. Getting feedback from others is very important in this hobby, and in life. Being able to graciously accept it is necessary. This goes for us as devs, and those opening issues here. This doesn't mean either party must follow said advice, of course. It's about the attitude around it.

For example, the OP's attitude here is abundant for all readers. On the other hand, we listened to the feedback given by OP and implemented "raw_unsync" to help APT HT users. Thanks to listening to feedback, this option is now available to all.

### Attacks against the devs
I was not "mean", and @Aang23 was not arguing about the scope of the project. In both situations we were giving feedback to a new user (see above). OP's assertions were simple ad hominem attacks, and are not productive.

Furthermore, an irrelevant issue was brought up (the one about selecting the wrong pipeline causing a crash). Even here, we try to fix these crashes when they come up, but selecting the wrong option still won't work - so we tell people they selected the wrong pipeline when they did. The other poster accepted the feedback, as is appropriate. There was no reason to bring it up again here, other than to bolster attacks against the devs.

## In Conclusion
There is a professional attitude we hold here, and we expect visitors to do the same. As you can see here, we will still help you even if you don't - _but please, don't be that guy._ Everyone will see it for what it is.

closed 2024-03-23 by JVital2013