SatDump/SatDump #634: NOAA APT misaligned composite image

Kappa971, 2024-02-17:
**Description of the issue**
Hi, I'm trying to receive NOAA satellites. I captured this (not perfect) image from NOAA 19 but when I generate the composite image, everything is misaligned. I received other images where this did not happen. 
This also happened to me in Wxtoimg, but there is an option to move the map overlay up or down, but there is no option to change the "angle" of the map (I don't know the technical term), it is sometimes wrong too.
Maybe an option for a manual adjustment could be the solution, I don't know.

**Hardware (SDR/PC/OS)**
Windows 10
RTL-SDR Blog V3

**Version (Eg, 1.0.0, CI Build 171)**
Satdump v1.1.3

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**

[image]

Aang23, 2024-02-17:
At first glance and from what you said it is probably due to your PC's clock not being quite on time, eg a few seconds off. Perhaps try to enable NTP?

As for the ""horizontal"" offset, this is something that doesn't need to be re-tuned normally. Though if it's NOAA-18, the gyros onboard are toast leading to high errors. But these are still small enough on APT in principle. 

Looking at the data I had from my station this morning, it doesn't seem like any of the sats changed its offsets much at all. So I definitely think it is either old TLEs or a time issue. If that doesn't solve it I'll ask for more info.

Kappa971, 2024-02-17:
> At first glance and from what you said it is probably due to your PC's clock not being quite on time, eg a few seconds off. Perhaps try to enable NTP?

It's possible, but I think Windows automatically synchronizes the time with a Microsoft server (maybe is it inaccurate?), I don't know what I could do to improve the accuracy. Anyway, it only happened with this image, or at least with this image the misalignment is very obvious.

> As for the ""horizontal"" offset, this is something that doesn't need to be re-tuned normally. Though if it's NOAA-18, the gyros onboard are toast leading to high errors. But these are still small enough on APT in principle.

With Wxtoimg I had episodes where the horizontal offset was incorrect, not only with NOAA 18, but looking more closely at this image, I notice that this is not the case.

> Looking at the data I had from my station this morning, it doesn't seem like any of the sats changed its offsets much at all. So I definitely think it is either old TLEs or a time issue. If that doesn't solve it I'll ask for more info.

I think TLE are up to date (I didn't change the default SatDump setting) and yes it might be a time issue but I don't know how to improve the situation.
An option to move the map overlay would perhaps help when this, for some reason, happens. It would probably be better to ask people much more experienced than me if this happened to them too (I'm using SatDump for a few days and I'm not a radio expert).

I'll keep trying over the next few days and see if it happens again.

RAD750, 2024-02-18:
Horizontal deviations are expected. The NOAA POES aren't the most stable of satellites, and on all except NOAA 15 they are operating in RGYRO mode due to 3 (NOAA 18) or 2 (NOAA 19) laser gyro failures.

The vertical offset points to a small (10 seconds tops?) timing offset. What are you using to record? Are you using SatDump , or another program, and you're supplying the wav?

Kappa971, 2024-02-18:
> Horizontal deviations are expected. The NOAA POES aren't the most stable of satellites, and on all except NOAA 15 they are operating in RGYRO mode due to 3 (NOAA 18) or 2 (NOAA 19) laser gyro [...]

I used Satdump for everything, but only in this case I had this problem (or at least in such a noticeable way) so I think it's either Windows or the satellite's fault. In any case, the image is irrecoverable unless left in black and white.

I think that, as already said, something to move the map/colors overlay could help in these unpredictable cases, but obviously it's just a suggestion (and I don't know how much work such a feature would require, nor if it's possible to do).

RAD750, 2024-02-18:
It's not unrecoverable!
If you have the WAV, you can reprocess it by specifying a slightly different timestamp. If not, we need to do a little bit more work, but it's possible.

Kappa971, 2024-02-18:
> It's not unrecoverable! If you have the WAV, you can reprocess it by specifying a slightly different timestamp. If not, we need to do a little bit more work, but it's possible.

I didn't save the recording, from now on I will

Kappa971, 2024-02-20:
I captured more images, and the problem described in the first post never recurred, so I can assume that it was a malfunction of my system (or the satellite).
However the problem of the horizontal deviation of the map/color overlay is always present, so I imagine that to have a 100% correct image, SatDump should allow you to manually correct this:

NOAA 19
[image]

NOAA 18
[image]

Kappa971, 2024-03-03:
I think I can close this issue as it was an exceptional case. 
However if someday an option to move the map/color overlay was added, it would be useful. Thank you

closed 2024-03-03 by Kappa971