SatDump/SatDump #635: Assertion `g.ItemFlagsStack.Size > 1' failed.

krakenrf, 2024-02-19:
Not sure what this crash is about, but it occurred during a Meteor HRPT pass. Meteor HRPT is usually fine for me.

Orange Pi 5, Ubuntu 22.04. Satdump-v1.1.4-fe4af748

```
[04:19:48 - 19/02/2024] (I) Starting processing pipeline meteor_hrpt...
[04:19:48 - 19/02/2024] (D) Input file (cadu) : /home/orangepi/nas/Discovery_Dish/live_output/2024-02-19_04-04_meteor_hrpt_1.7 GHz/meteor_hrpt.cadu
[04:19:48 - 19/02/2024] (D) Output file : /home/orangepi/nas/Discovery_Dish/live_output/2024-02-19_04-04_meteor_hrpt_1.7 GHz
[04:19:48 - 19/02/2024] (T) Update pass trajectory...
[04:19:48 - 19/02/2024] (D) Starting meteor_hrpt
[04:19:48 - 19/02/2024] (I) Data is already at level baseband, skipping
[04:19:48 - 19/02/2024] (I) Data is already at level soft, skipping
[04:19:48 - 19/02/2024] (I) Data is already at level cadu, skipping
[04:19:48 - 19/02/2024] (I) Processing data to level products
[04:19:48 - 19/02/2024] (D) Parameters :
[04:19:48 - 19/02/2024] (D)    - baseband_format : "f32"
[04:19:48 - 19/02/2024] (D)    - buffer_size : 1000000
[04:19:48 - 19/02/2024] (D)    - dc_block : true
[04:19:48 - 19/02/2024] (D)    - freq_shift : 0
[04:19:48 - 19/02/2024] (D)    - iq_swap : false
[04:19:48 - 19/02/2024] (D)    - samplerate : 2400000
[04:19:48 - 19/02/2024] (D)    - start_timestamp : 1708315458.0
[04:19:48 - 19/02/2024] (I) Using input frames /home/orangepi/nas/Discovery_Dish/live_output/2024-02-19_04-04_meteor_hrpt_1.7 GHz/meteor_hrpt.cadu
satdump-ui: /home/orangepi/SatDump/src-core/imgui/imgui.cpp:7211: void ImGui::PopItemFlag(): Assertion `g.ItemFlagsStack.Size > 1' failed.
Aborted (core dumped
```

JVital2013, 2024-02-19:
Hmm, looks like a bug in the UI - were you using the scheduler? If so, was it in multi mode?

krakenrf, 2024-02-20:
GUI and scheduler was in use, but no multi-mode. This is what I'm tracking:

[image]

Aang23, 2024-02-20:
I see. I'll try to reproduce on my end, though it might require leaving it running on an actual station which may take a bit more time.

Actually just wondering, why are you not tracking MetOp-B/C?

krakenrf, 2024-02-20:
> Actually just wondering, why are you not tracking MetOp-B/C?

The SNR is not high enough with an RTL-SDR to decode anything because of the half bandwidth.

If I use an Airspy it's fine, but I'm currently testing RTL-SDR.

JVital2013, 2024-02-22:
I believe this is fixed - could you update to the latest commit and try again?

krakenrf, 2024-02-23:
Thanks, will do. Will run it for a few days. I actually haven't seen this bug occur again for the past few days on the older commit version.

krakenrf, 2024-02-27:
Seems fixed.

closed 2024-02-27 by krakenrf