SatDump/SatDump #637: Create a type anchor file .vrt to place the image on the map.

stil-rambler, 2024-02-20:
Hi!
I haven't been using your program for long, but I really like it. But I would like to ask about adding one feature.
I would like to place pictures on a map and also use them in various mapping services. A binding file is required for this. I wrote my own generation program based on various libraries, but it works with glitches. I would like to know if there are any plans to create georeference files to improve the work with images in mapping systems. At least 4 extreme points in any format would be enough for me. For example, it would be possible to record it in a dataset.json.
With Noaa there are not so many problems, but with Meteor 2.3 it’s not possible to make a binding.

Aang23, 2024-02-20:
The ability to export in GeoTIFF format is planned - would that work for you too?

stil-rambler, 2024-02-20:
Yes!

closed 2024-02-20 by stil-rambler