SatDump/SatDump #638: Crash when loading TLE

Keeperbe, 2024-02-21:
**Description of the issue**
Logs shows "0 TLEs loaded!" and then the software crash.
I also can not find satdump_tles.txt in %appdata%/satdump

**Hardware (SDR/PC/OS)**
PC windows10

**Version (Eg, 1.0.0, CI Build 171)**
Release 1.1.4

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)
[01:52:22 - 21/02/2024] (E) Usage : C:\Program Files\SatDump\bin\satdump-ui.exe [downlink] [input_level] [input_file] [output_file_or_directory] [additional options as required]
[01:52:22 - 21/02/2024] (E) Extra options (examples. Any parameter used in modules can be used here) :
[01:52:22 - 21/02/2024] (E)  --samplerate [baseband_samplerate] --baseband_format [f32/i16/i8/w8] --dc_block --iq_swap
[01:52:22 - 21/02/2024] (D) Starting with OpenGL 4.6.0 NVIDIA 425.25
[01:52:22 - 21/02/2024] (D) Max texture size: 32768
[01:52:22 - 21/02/2024] (I)    _____       __  ____                      
[01:52:22 - 21/02/2024] (I)   / ___/____ _/ /_/ __ \__  ______ ___  ____ 
[01:52:22 - 21/02/2024] (I)   \__ \/ __ `/ __/ / / / / / / __ `__ \/ __ \
[01:52:22 - 21/02/2024] (I)  ___/ / /_/ / /_/ /_/ / /_/ / / / / / / /_/ /
[01:52:22 - 21/02/2024] (I) /____/\__,_/\__/_____/\__,_/_/ /_/ /_/ .___/ 
[01:52:22 - 21/02/2024] (I)                                     /_/      
[01:52:22 - 21/02/2024] (I) Starting SatDump v1.1.4
[01:52:22 - 21/02/2024] (I) 
[01:52:22 - 21/02/2024] (I) Loading config C:\Program Files\SatDump\share\satdump\satdump_cfg.json
[01:52:22 - 21/02/2024] (W) No user configuration found! Keeping defaults.
[01:52:22 - 21/02/2024] (I) Saving user config at C:\Users\joe\AppData\Roaming/satdump/settings.json
[01:52:22 - 21/02/2024] (I) Loading plugins from C:\Program Files\SatDump\lib\satdump\plugins
[01:52:22 - 21/02/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\aaronia_sdr_support.dll...
[01:52:22 - 21/02/2024] (E) Error loading C:\Program Files\SatDump\lib\satdump\plugins\aaronia_sdr_support.dll! Error : "C:\Program Files\SatDump\lib\satdump\plugins\aaronia_sdr_support.dll": 找不到指定的模块。
[01:52:22 - 21/02/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\aim_support.dll...
[01:52:22 - 21/02/2024] (T) Plugin aim_support loaded!
[01:52:22 - 21/02/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\airspyhf_sdr_support.dll...
[01:52:22 - 21/02/2024] (T) Plugin airspyhf_sdr_support loaded!
[01:52:22 - 21/02/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\airspy_sdr_support.dll...
[01:52:22 - 21/02/2024] (T) Plugin airspy_sdr_support loaded!
[01:52:22 - 21/02/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\analog_support.dll...
[01:52:22 - 21/02/2024] (T) Plugin analog_support loaded!
[01:52:22 - 21/02/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\bladerf_sdr_support.dll...
[01:52:22 - 21/02/2024] (T) Plugin bladerf_sdr_support loaded!
[01:52:22 - 21/02/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\bluewalker3_support.dll...
[01:52:22 - 21/02/2024] (T) Plugin bluewalker3_support loaded!
[01:52:22 - 21/02/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\cluster_support.dll...
[01:52:22 - 21/02/2024] (T) Plugin cluster_support loaded!
[01:52:22 - 21/02/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\cubesat_support.dll...
[01:52:22 - 21/02/2024] (T) Plugin cubesat_support loaded!
[01:52:22 - 21/02/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\dmsp_support.dll...
[01:52:22 - 21/02/2024] (T) Plugin dmsp_support loaded!
[01:52:22 - 21/02/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\dscovr_support.dll...
[01:52:22 - 21/02/2024] (T) Plugin dscovr_support loaded!
[01:52:22 - 21/02/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\dvb_support.dll...
[01:52:22 - 21/02/2024] (T) Plugin dvb_support loaded!
[01:52:22 - 21/02/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\elektro_arktika_support.dll...
[01:52:22 - 21/02/2024] (T) Plugin elektro_arktika_support loaded!
[01:52:22 - 21/02/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\eos_support.dll...
[01:52:22 - 21/02/2024] (T) Plugin eos_support loaded!
[01:52:22 - 21/02/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\fengyun2_support.dll...
[01:52:22 - 21/02/2024] (T) Plugin fengyun2_support loaded!
[01:52:22 - 21/02/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\fengyun3_support.dll...
[01:52:22 - 21/02/2024] (T) Plugin fengyun3_support loaded!
[01:52:22 - 21/02/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\fengyun4_support.dll...
[01:52:22 - 21/02/2024] (T) Plugin fenyun4_support loaded!
[01:52:22 - 21/02/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\gcom_support.dll...
[01:52:22 - 21/02/2024] (T) Plugin gcom_support loaded!
[01:52:22 - 21/02/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\geonetcast_support.dll...
[01:52:22 - 21/02/2024] (T) Plugin geonetcast_support loaded!
[01:52:22 - 21/02/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\gk2a_support.dll...
[01:52:22 - 21/02/2024] (T) Plugin gk2a_support loaded!
[01:52:22 - 21/02/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\goes_support.dll...
[01:52:22 - 21/02/2024] (T) Plugin goes_support loaded!
[... 475 lines cut ...]
[01:52:23 - 21/02/2024] (D)  - stereo_hr
[01:52:23 - 21/02/2024] (D)  - stereo_lr
[01:52:23 - 21/02/2024] (D)  - stratosat_tk1_uhf
[01:52:23 - 21/02/2024] (D)  - npp_hrd
[01:52:23 - 21/02/2024] (D)  - swot_tlm
[01:52:23 - 21/02/2024] (D)  - syracuse3b_tlm
[01:52:23 - 21/02/2024] (D)  - terra_db
[01:52:23 - 21/02/2024] (D)  - test_custom
[01:52:23 - 21/02/2024] (D)  - tiahne1_tlm
[01:52:23 - 21/02/2024] (D)  - tianwen1_link
[01:52:23 - 21/02/2024] (D)  - timed_dump
[01:52:23 - 21/02/2024] (D)  - tubin_x_dump
[01:52:23 - 21/02/2024] (D)  - umka_1_dump
[01:52:23 - 21/02/2024] (D)  - viasat3_tlm_12200
[01:52:23 - 21/02/2024] (D)  - viasat3_tlm
[01:52:23 - 21/02/2024] (D)  - youthsat_dump
[01:52:23 - 21/02/2024] (D)  - yunhai_ahrpt
[01:52:23 - 21/02/2024] (I) Found OpenCL Devices (2) :
[01:52:23 - 21/02/2024] (D)  - Intel(R) UHD Graphics 620
[01:52:23 - 21/02/2024] (D)  - GeForce MX250
[01:52:23 - 21/02/2024] (I) Loading TLEs from C:\Users\joe\AppData\Roaming/satdump/satdump_tles.txt
[01:52:23 - 21/02/2024] (I) 0 TLEs loaded!

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**
[image]

JVital2013, 2024-02-21:
Hello!

The "0 TLEs loaded!" message is likely not a problem, as this is normal for a first launch. This crash is likely occurring while probing SDRs, which happens after that message.

Do you have any SDRs connected to your system?

Keeperbe, 2024-02-21:
> Hello!
> 
> The "0 TLEs loaded!" message is likely not a problem, as this is normal for a first launch. This crash is likely occurring while probing SDRs, which happens after that message.
> 
> Do you have any SDRs connected to your system?

I have a RTL2832SDR

Keeperbe, 2024-02-21:
> Hello!
> 
> The "0 TLEs loaded!" message is likely not a problem, as this is normal for a first launch. This crash is likely occurring while probing SDRs, which happens after that message.
> 
> Do you have any SDRs connected to your system?

I have a RTL2832SDR. However, the software crashes regardless of whether I connect the SDR to the computer or not.
Thanks for your help！

darayavahu, 2024-03-02:
The same problem.

darayavahu, 2024-03-02:
Thanks for you reply. I try what you wrote and remote_sdr_support.dll causing the problem, without this file satdump work.

JVital2013, 2024-03-02:
Thanks @darayavahu for the info! That gives us something to go off of. Do you have any Antiviruses or security tools installed on your PC?

Also, can you confirm @Keeperbe that removing `C:\Program Files\SatDump\lib\satdump\plugins\remote_sdr_support.dll` works around the issue for you as well?

darayavahu, 2024-03-02:
I use only windows defender.
Yes i can confirm that removing C:\Program Files\SatDump\lib\satdump\plugins\remote_sdr_support.dll works for me. Tested few minutes ago on noaa apt on rtlsdr. I use satdump for the first time.

Keeperbe, 2024-03-25:
> Thanks @darayavahu for the info! That gives us something to go off of. Do you have any Antiviruses or security tools installed on your PC?
> 
> Also, can you confirm @Keeperbe that removing `C:\Program Files\SatDump\lib\satdump\plugins\remote_sdr_support.dll` works around the issue for you as well?

It works！Thank you all！

JVital2013, 2024-03-25:
I'm glad deleting the file works for you, and that's a fine workaround!

Ultimately, I'd like to fix the remote SDR plugin for systems like yours though, so I re-opened the issue. But... I'm not sure why it would lock up a system. I'm a bit baffled, as it works fine on Windows 10 for me.

Do you have any Antiviruses, firewalls, or system proxies installed @Keeperbe?

Aang23, 2024-03-25:
That's very much a weird one... Where does it lock up at in the logs?

Keeperbe, 2024-03-25:
> I'm glad deleting the file works for you, and that's a fine workaround!
>
> Ultimately, I'd like to fix the remote SDR plugin for systems like yours though, so I re-opened the issue. But... I'm not [...]

I try to close my _[proxy software](https://github.com/LibNyanpasu/clash-nyanpasu)_ which runs http and socks proxy in ports 7890/7898/7899 and the problem disappear. There might be some kind of conflict between this software and SatDump. This problem also occurs on my other win10 computer.

JVital2013, 2024-03-25:
@Aang23 it happens right after loading TLEs, so it must be when querying SDRs (and doing the broadcast discovery).

Thanks @Keeperbe, its likely related to proxy software. We'll do some testing!

minexixi, 2024-04-03:
I have also encountered a similar problem. This happened when I first started and there was no response when loading Tles. I modified the port of Clash from 7890 to 7891, and then the software ran normally.

YD9CKH, 2024-04-08:
> @Aang23 it happens right after loading TLEs, so it must be when querying SDRs (and doing the broadcast discovery).
> 
> Thanks @Keeperbe, its likely related to proxy software. We'll do some testing!

Hi, im not using Proxy software but i face same issue. But before it runs normal. what is the conflict?

JVital2013, 2024-04-10:
Hello @Keeperbe, @darayavahu, @BI4PYM, @YD9CKH, and anyone else I might have missed: This issue should be fixed as of the latest nightly (1.20-b1d5bc3, 1.20-587d2ff, or newer). Could you please update, and let us know if the crash is fixed for you?

It appears the crash revolves around SatDump's use of UDP ports 4567 and 7890, which is needed for the remote server auto-discovery process. This fix will make remote sdr auto-discovery gracefully fail instead of crashing the program, if necessary. In these cases, you can still use Remote SDR functionality by directly specifying IP addresses in settings.

Keeperbe, 2024-04-14:
> Hello @Keeperbe, @darayavahu, @BI4PYM, @YD9CKH, and anyone else I might have missed: This issue should be fixed as of the latest nightly (1.20-b1d5bc3, 1.20-587d2ff, or newer). Could you please [...]

I tried to install SatDump v1.2.0-a35e713, but the problem still occurred.

ZHENG0119, 2024-09-03:
[image] 
Ask for Help