SatDump/SatDump #641: SatDump APK

ve3elb, 2024-02-23:
**Description of the issue**

SatDump won't show or save output directories path. I have uninstall, reboot and reinstall SatDump but same issue can assign or save output directories. 

**Hardware (SDR/PC/OS)**
Samsung Tab

**Version (Eg, 1.0.0, CI Build 171)**
SatDump 1.1.4

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**

[image]

JVital2013, 2024-06-16:
Which Samsung Galaxy Tab A do you have?

I just happened to pick up a Samsung Galaxy Tab A (8.0 inches, 2019, Wi-Fi - SM-T290) at a yard sale. I tested SatDump on stock Android 10 and 11 and the file browser works fine for me, and lets me assign output paths

Mraanderson, 2025-08-06:
This is working for me on Android 15 (Galaxy A21s).

Edit paths, save and restart.