SatDump/SatDump #642: Support SDR Play- RSP1B

pa0alw, 2024-02-24:
Hallo,
Should it possible to support in Satcom for the SDRplay RSP1 receiver in the near future?

Greetings

Walter

Aang23, 2024-02-25:
Yes.

pa0alw, 2024-02-25:
Thanks,
When i connect a RSP1B i get the message:
Device:Unknow RSP-2401017760
           This device is not supported yet,or perhaps a clone

JVital2013, 2024-03-04:
Hi @pa0alw,

Can you try this again with the latest nightly? Some code was recently merged to support the RSP1B

pa0alw, 2024-03-04:
Which version? I don't have a Linux computer,
I can test only the 1.1.4 windows version
Walter.

JVital2013, 2024-03-04:
Nightly builds are available here: https://github.com/SatDump/SatDump/releases/tag/nightly

pa0alw, 2024-03-05:
Thanks,
I install the latest Nightly version V1.2.0-324d226.
I see the correct device RSP1B 2401017760 in the recorder screen, started it but there is no reception and waterfall info.

Walter

pa0alw, 2024-03-12:
With the latest Nightly version V1.2.0-ff6ca6 still no reception with the RSP1B in the waterfall/panadpter screen
You can see that the waterfall is running,

Walter.

Aang23, 2024-05-14:
@pa0alw this should be fixed in Windows builds.

closed 2024-05-14 by Aang23