SatDump/SatDump #643: Option for GEOTIFF?

sattrackcam, 2024-02-25:
An option to save created images as GEOtiff (or jpg with world file), i.e. a georeferenced image, so that they can be directly read into a GIS program, would be very welcome.

LEDFlighter, 2024-02-25:
What would you want to do with the georeferenced image in a GIS program? Do you want to project it? There is already a "projection"-tab in SatDump.

JVital2013, 2024-02-25:
This functionality is in development on the `proj_geotiff` branch as we speak! It's not quite ready for use, however.

Duplicate of #298 - we'll update that issue once more concrete progress has been made.

closed 2024-02-25 by JVital2013

sattrackcam, 2024-02-25:
Thanks @JVital2013 JVital2013 for the info. Good to know it is worked on, and looking forward to implementation!

In answer to @LEDFlighter : it allows to combine the satellite imagery with data from other (external) sources, as well as generate animations.
A GIS (Geograpghical Information System) program has much more options than SatDump can deliver in terms of (external) data combination, projections available etcetera. For example, you could combine the image with pressure or temperature measurements at ground level from various weatherstations, or with weathermaps etcetera, aircraft flight routes and you name it.