SatDump/SatDump #644: Does  SatDump  have memory leaks?

sarbo, 2024-02-26:
Running   SatDump   for  three  days    and  my  available  memory   keeps  going  down.l  
Orange Pi 5 Plus  with   recommended   Ubuntu  install.....

**Version (Eg, 1.0.0, CI Build 171)**

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**

Aang23, 2024-02-26:
We'll need a few more details on what you are you running to be able to get an idea. Are you running in UI mode?

sarbo, 2024-02-26:
Yes, I am running in UI mode.

sarbo, 2024-02-26:
No, it is enabled.  I will disable it and watch my available memory. Thanks

Aang23, 2024-02-26:
Did you disable "Open in viewer after processing"? If not chances are the data is simply being opened and kept in memory in the viewer tab. If so, disabling the option should fix the issue.

@RAD750 do you mention this in your guide actually?

RAD750, 2024-02-27:
@Aang23 Sure did, but none of my guides cover GOES as it's outside my reception range.
[image]

krakenrf, 2024-02-27:
@Aang23 Perhaps viewer should only keep the last X passes open, and close the older ones?

JVital2013, 2024-03-02:
Hello everyone! It looks like there are two different issues here - let's address them both.

### Does SatDump have memory leaks (when using the GOES-R HRIT Decoder)
First, let's address the issue originally opened by @sarbo. While I won't affirm SatDump has no memory leaks (it's a big program!), I've done extensive testing of the GOES-R HRIT decoder and cannot identify any memory leaks in this part. The original idea to disable "Open in viewer after processing" has no effect here, as GOES HRIT cannot currently be opened in the viewer automatically.

**So, if it's not a leak, where is that memory growth you're seeing coming from?** In this case, it looks like you're not quite getting a complete signal (perhaps several CRC errors?), so several download-in-process images are kept in memory at once. In your screenshot there are 5 images currently held in memory, which comes out to about 125MB of additional memory by itself. Additionally, several memory pages are kept by the memory allocator for other things (LRIT files, DSP buffers, image preview textures, etc), as they are used over and over again.

This can add up over the first few hours, but this type of memory growth will level out after a 12 hours or so, and should not continue to increase - although it may fluctuate up and down a bit, depending on what's "in progress."

If your memory growth does not level off, let us know.

### OK, so what _should_ be done about products that open from the scheduler in the viewer?
The issue brought up by @krakenrf (and others) about products that are decoded by autotrack, then subsequently opened in the viewer, is a different issue - but it's valid, of course, and tantamount to a memory leak. While turning that option off in settings certainly works, I'm thinking something should probably be done here - but I'm not sure only keeping X passes in the viewer is a good option.

My primary concern is I don't want the viewer to close passes on me if I happen to be using them when they decide to auto-close, for example. Some other ideas I had (these are different ideas, not all to be done at once):

- Add a checkbox to the scheduler to control auto-open in viewer separately for the scheduler
- Add a checkbox to a specific satellite's live config to open it specifically in viewer after decoding
- Only open automatically in viewer if there are fewer than X products already open (not a big fan of this one)

Of course I'm open to whatever here - I'm not married to any of these ideas!

sarbo, 2024-03-02:
Thank you @JVital2013  for clearing this up.  It was confusing.  I ran GOES-HRIT today for six hours monitoring memory using Htop and Psensor, file attached.  I think the decreasing available memory reported by Psensor is incorrect.  I trust Htop more. As you said fluctuations in  memory used but stable.

[SatDumpMemUsageGoes-HRIT.txt](https://github.com/SatDump/SatDump/files/14470964/SatDumpMemUsageGoes-HRIT.txt)

JVital2013, 2024-03-02:
Yeah, the psensor output doesn't make a lot of sense - especially since it contradicts htop. The htop one lines up with what I'd expect. A longer test should confirm this. I'm glad I could clear things up for you!

krakenrf, 2024-03-03:
@JVital2013 how about a "keep open" checkbox in viewer for a pass you'd like to not close? In my opinion I would like to easily be able to view just the latest passes and care less about older passes over time.

Either that or is there not a way to show all passes in the list, but lazy-load images only when the user clicks on a pass? And flush them out or memory after some time, and/or when more than X number have been loaded?