SatDump/SatDump #646: Samplerate is Zero in CLI autotrack setup

var-IO-reeks, 2024-02-26:
**Description of the issue**
Despite using a valid value of 768kSPS for the Samplerate of the Airspy HF+ that is running as a receiver in my automated station setup, the log output says
```
Feb 26 20:03:06 server satdump[1234567]: [19:03:05 - 26/02/2024] (E) Samplerate not set!
...
Feb 26 20:03:06 server satdump[1234567]: [19:03:05 - 26/02/2024] (E) Could not start recorder/processor since the source could not be started!

```
and the WebGUI also says that the samplerate is zero and the system is not started,
[image]

despite the usage of an autotrack_config.json-file that worked before and begins like this:
```
{
    "parameters": {
        "source": "spyserver",
        "ip_address": "192.168.1.3",
        "samplerate": 768000,
        "port": "5555",
        "bit_depth": 32,
        "gain": 0,
        "digital_gain": 0,
        "initial_frequency": 137.64e6,
        "fft_enable": true,
        "fft_size": 65536, 
        "fft_rate": 30,
        "fft_avgn": 1
    },
    "finish_processing": true,
...
```
A connection to the same Spyserver with the GUI version of satdump from two different computers works flawlessly.

**Hardware (SDR/PC/OS)**
Airspy HF+ Discovery connected to a Raspberry pi 3B+ running SpyServer, being received from another linux server that is running the CLI version of Satdump, where the command
```
satdump autotack autotrack_config.json
```
is running as a background service.

**Version (Eg, 1.0.0, CI Build 171)**
v1.1.5-91be8ae8, but that issue has existed since mid january

After doing some additional tests, I think the issue could also be with the connetion to the spyserver itself, since running the script for example with a wrong port number, the WebGUI still looks unchanged and and the CLI output doesent look all that wrong either.

JVital2013, 2024-03-03:
Hi @var-IO-reeks,

This should be fixed as of the latest commit. If you still have issues after updating, let us know!

closed 2024-03-03 by JVital2013

var-IO-reeks, 2024-03-03:
Yes, the Spyserver setup is indeed working again, big thanks for fixing!