SatDump/SatDump #648: Forum? Getting help?

mistergitj, 2024-02-28:
I am trying to get traction with SatDump. Where is a good place to do this? I was thinking, at least until I get some antennae in place, I would DL some test files and try viewing them. I have found some files, but I receive only errors when trying to view them. Thanks much. John.

Aang23, 2024-02-28:
If you want, we have a Matrix space for SatDump ( https://matrix.to/#/#satdump:altillimity.com ) which serves for most of these purposes. If needed I can easily provide some test files and similar.

mistergitj, 2024-02-28:
Thanks, Aang23. I am not  familiar with Matrix but am in the process of getting an account. Thanks and enjoy, John.

mistergitj, 2024-02-29:
Aang23, I am not knowledge enough with Matrix. Please provide me a couple of image test files for SatDump. Thanks, John.

Aang23, 2024-03-01:
Which satellite do you want samples of?

mistergitj, 2024-03-01:
Thanks, Aang23. Probably NOAA. I have an antenna which should work. Thanks much, John.

mpentler, 2024-03-09:
> Aang23, I am not knowledge enough with Matrix. Please provide me a couple of image test files for SatDump. Thanks, John.

Can we help with Matrix at all? Did you get as far as making an account?