SatDump/SatDump #649: Show local time in Schedule for autotrack

Glassnost, 2024-02-29:
**Description of the issue**
The schedule for Autotrack in satdump-ui does not show up in local time zone
In my case it´s CET - 1h

Perhaps not a "bug", but a configuration purpose ?

**Hardware (SDR/PC/OS)**
Debian, 
RTL-SDR V4

**Version (Eg, 1.0.0, CI Build 171)**
currently in use : v1.1.5-87446a11 
**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)
/
**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**
[image]

Aang23, 2024-03-04:
It was intended for all timestamps to be in UTC for the sake of simplicity. While it's possible to do thing in local time too, it's also quite a bit more work. I don't know if it's really worth it. Most other SW will also be in UTC.

chanrobi, 2024-05-01:
Incredibly short sighted to not even ALLOW the option of Local time. When I or anyone else is figuring out satellite passes, eventually you have to convert to local time. 

NOBODY uses UTC on a daily basis. The clock on my alarm, my phone, my watch are not in UTC nor do you think in UTC. 

So to get a huge list of satellite passes ONLY in UTC because someone thought it'd be "too much work" and "all other software is in UTC" is honestly, ridiculous.

Neither Gpredict or WxtoImg did UTC only