SatDump/SatDump #652: Meteor-m2-4 Map overlay

pdxh0b0, 2024-03-01:
It sits at "update image..." indefinitely 

Nooelec v5, Microsoft surface laptop 

Nightly & current stable 

Cadu
https://drive.google.com/file/d/1tZ3RCVlC4dy7S32wEqez_mkYUQaeJEOL/view?usp=drivesdk

Aang23, 2024-03-02:
Was this using a 1.2.0 nightly? Support for M2-4 overlays etc was added just yesterday.

pdxh0b0, 2024-03-02:
1.2.0-66fff30 currently installed

Downloaded and installed 03/01 at 3pm pst. Just now uninstalled/reinstalled same result

Instead of using a cadu created by another program, overlays working if I process the baseband file with satdump

Meteor-m2-3 cadu outputs by meteordemod work no issues in satdump, just the Meteor-m2-4 cadu...

Aang23, 2024-03-02:
Well if another software's CADU is not working when processing in SatDump works fine, this is not a SatDump problem...

Aang23, 2024-03-02:
As a note, I've notified @Digitelektro about this. I'll close this for now - unless it turns out there's more on this end.

pdxh0b0, 2024-03-02:
Ty

closed 2024-03-02 by Aang23