SatDump/SatDump #653: SatDump 1.1.4 and nightly 1.4 crash

RubbaCippi, 2024-03-02:
After connecting to SpyServer (working) clicking on Start crash the application.
For more detail take a look at the attached screen recording and to the crash log.
The crash is fully repeatable.

MacBook Pro 14" 2021 M1Max 64 Giga ram

**Version 1.1.4 and 1.4

**Logs after the crash** 

[SatDump 1.2.0 nightly crash.txt](https://github.com/SatDump/SatDump/files/14468716/SatDump.1.2.0.nightly.crash.txt)
[satdump-20240302T102428.log](https://github.com/SatDump/SatDump/files/14468722/satdump-20240302T102428.log)

**Other info (Eg, Screenshots) 

https://github.com/SatDump/SatDump/assets/155834790/e9745d81-c8dc-43b8-9b08-3181b430c091

RubbaCippi, 2024-03-02:
> I suspect this is the same bug as #646 - something with the SpyServer plugin. I'll look into it later - for now I'll keep both issues open in case they are indeed different.

Thankyou!

I look forward for the fix :-)

RubbaCippi, 2024-03-02:
Hi Jamie,
I downloaded the Windows version, and it work.
I can successfully connect to my spy server and start the reception.
No satellites, actually…

I run SatDump under Windows11 ARM, which run on Parallel in MacBook Pro 14” M1Max

Hope this help

JVital2013, 2024-03-02:
Thanks for all the info! After some investigation I believe this is different from the other issue.

Your problem, though, should now be fixed - can you try the latest nightly from https://github.com/SatDump/SatDump/releases/tag/nightly and let us know how it works for you?

RubbaCippi, 2024-03-03:
Great!
SatDump 1.2.0-b396d90 don’t crash.
Now I’ll wait to receive a transmission or to find a I/Q recording to do some experience with it.

Thank you for the support.
Roberto

closed 2024-03-03 by JVital2013