SatDump/SatDump #654: Command line arguments for file input in satdump live CLI?

nils10, 2024-03-02:
Hi,

In the UI its possible to select in the recordings tab a file input to replay a recording (as if it was a live pass). This is quite useful for testing.

Is it also possible to use an IQ file as input for satdump CLI version? What would the cmd line parameters be in that case?

I was trying along the lines of `satdump live goes_grb goes_grb_live_test --source file "recording.iq" --frequency 1686.6e6 --samplerate 10000000 --server_address 0.0.0.0 --server_port 5004 --http_server 0.0.0.0:8081` but I still seem to be missing something.

JVital2013, 2024-03-02:
Hello @nils10,

Yes, this is possible - normally I'd recommend using the offline processing command line arguments, which would go something like this:

```bash
satdump goes_grb baseband recording.iq goes_grb_test --samplerate 10000000 --baseband_format s16 #Or whatever format you're using
```

It sounds like you specifically want to do live playback with a file from the CLI, though. You can do this too, like this:

```bash
satdump live goes_grb goes_grb_live_test --source file --file_path recording.iq --frequency 1686.6e6 --samplerate 10000000 --baseband_type s16 #Or whatever format you're using
```

This currently is not documented, other than some discussion in #447 - the list of parameters is in one of my comments further down in the issue.  Its a rather niche use case - but I hope this helps!

nils10, 2024-03-02:
Hi @JVital2013,

Thats great, thank you! For completeness, I was wondering: Are there also CLI arguments for enabling the "Fast" replay and iq_swap?

JVital2013, 2024-03-02:
Not for fast replay unfortunately, but yes `iq_swap` works

JVital2013, 2024-03-04:
Since this has come up a few times now, I've slated the info to be added to https://docs.satdump.org/sdr_options.html on next site update.

[image]

closed 2024-03-04 by JVital2013