SatDump/SatDump #655: Regarding the software crash phenomenon during METEOR M2-4 reception

X-MQSI, 2024-03-04:
**Description of the issue**
Regarding the software crash phenomenon during METEOR M2-4 reception

**Hardware (SDR/PC/OS)**
SDR: RTL-SDR V4
PC:  Desktop computer assembled by my self
OS: Windows 10 Professional Workstation

**Version (Eg, 1.0.0, CI Build 171)**
The Latest Version 1.1.4 in Release

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)

[00:38:47 - 04/03/2024] (I) Writing images.... (Can take a while)
[00:38:47 - 04/03/2024] (T) MSU-MR ID -1
[00:38:47 - 04/03/2024] (I) ----------- Satellite
[00:38:47 - 04/03/2024] (I) NORAD : 0
[00:38:47 - 04/03/2024] (I) Name  : Unknown Meteor

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**
None

X-MQSI, 2024-03-04:
OK，the software crashed again while I was receiving NOAA-15。

[12:39:52 - 04/03/2024] (C) Couldn't calibrate image with wedges! Likely no valid wedge was identified!
[12:39:52 - 04/03/2024] (I) Saving C:\Users\MQSI\Documents/live_output/2024-03-04_12-26_noaa_apt_137.62 MHz/raw_sync.png...
[12:39:52 - 04/03/2024] (I) ----------- Satellite
[12:39:52 - 04/03/2024] (I) NORAD : 25338
[12:39:52 - 04/03/2024] (I) Name  : NOAA-15
[12:39:52 - 04/03/2024] (W) Congratulations for receiving NOAA 15 on APT! It has been 9426 days, 20 hours, 34 minutes and 3 seconds since it has been launched.
[12:39:52 - 04/03/2024] (I) ----------- AVHRR
[12:39:52 - 04/03/2024] (I) Lines : 1653
[12:39:52 - 04/03/2024] (W) Identified channels (debug): A: -1 A1: -1 B: -1
[12:39:52 - 04/03/2024] (I) Saving C:\Users\MQSI\Documents/live_output/2024-03-04_12-26_noaa_apt_137.62 MHz/APT-B.png...
[12:39:52 - 04/03/2024] (I) Saving C:\Users\MQSI\Documents/live_output/2024-03-04_12-26_noaa_apt_137.62 MHz/APT-A.png...
[12:39:52 - 04/03/2024] (I) Starting processing pipeline noaa_apt...
[12:39:52 - 04/03/2024] (D) Input file (products) : C:\Users\MQSI\Documents/live_output/2024-03-04_12-26_noaa_apt_137.62 MHz/dataset.json
[12:39:52 - 04/03/2024] (D) Output file : C:\Users\MQSI\Documents/live_output/2024-03-04_12-26_noaa_apt_137.62 MHz
[12:39:52 - 04/03/2024] (D) Starting noaa_apt
[12:39:52 - 04/03/2024] (I) Data is already at level baseband, skipping
[12:39:52 - 04/03/2024] (I) Data is already at level audio_wav, skipping
[12:39:52 - 04/03/2024] (I) Data is already at level products, skipping
[12:39:52 - 04/03/2024] (D) Products processing is enabled! Starting processing module.
[12:39:52 - 04/03/2024] (I) C:\Users\MQSI\Documents/live_output/2024-03-04_12-26_noaa_apt_137.62 MHz/.
[12:39:52 - 04/03/2024] (I) C:\Users\MQSI\Documents/live_output/2024-03-04_12-26_noaa_apt_137.62 MHz/./product.cbor
[12:39:52 - 04/03/2024] (I) C:\Users\MQSI\Documents/live_output/2024-03-04_12-26_noaa_apt_137.62 MHz/./product.cbor
[12:39:52 - 04/03/2024] (I) Loading APT-A.png
[12:39:52 - 04/03/2024] (I) Waiting for the thread...
[12:39:52 - 04/03/2024] (I) Loading APT-B.png
[12:39:52 - 04/03/2024] (D) Skipping 221 as it can't be made!
[12:39:52 - 04/03/2024] (D) Skipping 3a21 as it can't be made!
[12:39:52 - 04/03/2024] (D) Skipping 4/221 Merge as it can't be made!
[12:39:52 - 04/03/2024] (D) Skipping NOAA Natural Color as it can't be made!
[12:39:52 - 04/03/2024] (D) Skipping Natural Color as it can't be made!
[12:39:52 - 04/03/2024] (D) Skipping 543b as it can't be made!
[12:39:52 - 04/03/2024] (D) Skipping 3b45 as it can't be made!
[12:39:52 - 04/03/2024] (D) Skipping 124 as it can't be made!
[12:39:52 - 04/03/2024] (D) Skipping Thermal Channel as it can't be made!
[12:39:52 - 04/03/2024] (D) Skipping Night Microphysics as it can't be made!
[12:39:52 - 04/03/2024] (D) Skipping Day Microphysics as it can't be made!
[12:39:52 - 04/03/2024] (D) Skipping MCIR as it can't be made!
[12:39:52 - 04/03/2024] (D) Skipping MSA (channel 1) as it can't be made!
[12:39:52 - 04/03/2024] (D) Skipping MSA (channel 2) as it can't be made!
[12:39:52 - 04/03/2024] (D) Skipping NO enhancement as it can't be made!
[12:39:52 - 04/03/2024] (D) Composite needs channel cha
[12:39:52 - 04/03/2024] (T) Max Total Width 909
[12:39:52 - 04/03/2024] (T) Max Total Used  909
[12:39:52 - 04/03/2024] (T) Offset for cha is 0
[12:39:52 - 04/03/2024] (I) Saving C:\Users\MQSI\Documents/live_output/2024-03-04_12-26_noaa_apt_137.62 MHz/./avhrr_3_APT_channel_A.png...
[12:39:52 - 04/03/2024] (I) Thread stopped
[12:39:53 - 04/03/2024] (I) Saving user config at C:\Users\MQSI\AppData\Roaming/satdump/settings.json
[12:39:53 - 04/03/2024] (D) Composite needs channel chb
[12:39:53 - 04/03/2024] (T) Max Total Width 909
[12:39:53 - 04/03/2024] (T) Max Total Used  909
[12:39:53 - 04/03/2024] (T) Offset for chb is 0
[12:39:53 - 04/03/2024] (I) Saving user config at C:\Users\MQSI\AppData\Roaming/satdump/settings.json
[12:39:53 - 04/03/2024] (T) Updating elements...
[12:39:53 - 04/03/2024] (T) Update pass trajectory...
[12:39:53 - 04/03/2024] (I) Saving C:\Users\MQSI\Documents/live_output/2024-03-04_12-26_noaa_apt_137.62 MHz/./avhrr_3_APT_channel_B.png...
[12:39:53 - 04/03/2024] (D) Composite needs channel chb
[12:39:53 - 04/03/2024] (T) Max Total Width 909
[12:39:53 - 04/03/2024] (T) Max Total Used  909
[12:39:53 - 04/03/2024] (T) Offset for chb is 0
[12:39:53 - 04/03/2024] (D) Filtering timestamps...

JVital2013, 2024-03-04:
Hello!

SatDump 1.1.4 does not support METEOR M2-4, and it was released before the satellite launched. Please try the nightly available at https://github.com/SatDump/SatDump/releases/tag/nightly, and let us know if either of these crashes continue after updating

Aang23, 2024-03-04:
Well to be fair 1.1.4 should not crash on M2-4 either.

X-MQSI, 2024-03-07:
> Hello!
> 
> SatDump 1.1.4 does not support METEOR M2-4, as it was released before the satellite launched. Please try the nightly available at https://github.com/SatDump/SatDump/releases/tag/nightly, and let us know if either of these crashes continue after updating

Thank you, I downloaded the night version of the auto-build, and it works well.😎

JVital2013, 2024-03-07:
Glad to hear it! If it happens again let us know

closed 2024-03-07 by JVital2013