SatDump/SatDump #656: Meteor M2-4 HRPT Crash

krakenrf, 2024-03-04:
Getting a crash when receiving Meteor M2-4 HRPT. Like before it might be the blocked pass/empty data problem, or recently just because M2-4 was turned off.

Running satdump-ui 1.2.0-66fff300 on an Orange Pi 5 with an Airspy dongle.

I recorded a raw IQ file too, but I believe it's just all noise.

```
[13:38:25 - 04/03/2024] (T) Update pass trajectory...
[15:03:40 - 04/03/2024] (C) AOS!!!!!!!!!!!!!! 59051
[15:03:40 - 04/03/2024] (D) Set Airspy frequency to 1700000000
[15:03:40 - 04/03/2024] (T) Start pipeline...
[15:03:40 - 04/03/2024] (I) Generated folder name : /home/orangepi/nas/Discovery_Dish/live_output/2024-03-04_15-03_meteor_hrpt_1.7 GHz
[15:03:40 - 04/03/2024] (D) Parameters :
[15:03:40 - 04/03/2024] (D)    - baseband_format : "f32"
[15:03:40 - 04/03/2024] (D)    - buffer_size : 1000000
[15:03:40 - 04/03/2024] (D)    - clock_omega_relative_limit : 0.001
[15:03:40 - 04/03/2024] (D)    - dc_block : true
[15:03:40 - 04/03/2024] (D)    - freq_shift : 0
[15:03:40 - 04/03/2024] (D)    - iq_swap : false
[15:03:40 - 04/03/2024] (D)    - pll_bw : 0.01
[15:03:40 - 04/03/2024] (D)    - rrc_alpha : 0.5
[15:03:40 - 04/03/2024] (D)    - samplerate : 10000000
[15:03:40 - 04/03/2024] (D)    - start_timestamp : 1709564620.0
[15:03:40 - 04/03/2024] (D)    - symbolrate : 665400
[15:03:40 - 04/03/2024] (I) Module pm_demod
[15:03:40 - 04/03/2024] (D) Parameters :
[15:03:40 - 04/03/2024] (D)    - baseband_format : "f32"
[15:03:40 - 04/03/2024] (D)    - buffer_size : 1000000
[15:03:40 - 04/03/2024] (D)    - clock_omega_relative_limit : 0.001
[15:03:40 - 04/03/2024] (D)    - dc_block : true
[15:03:40 - 04/03/2024] (D)    - freq_shift : 0
[15:03:40 - 04/03/2024] (D)    - iq_swap : false
[15:03:40 - 04/03/2024] (D)    - pll_bw : 0.01
[15:03:40 - 04/03/2024] (D)    - rrc_alpha : 0.5
[15:03:40 - 04/03/2024] (D)    - samplerate : 10000000
[15:03:40 - 04/03/2024] (D)    - start_timestamp : 1709564620.0
[15:03:40 - 04/03/2024] (D)    - symbolrate : 665400
[15:03:40 - 04/03/2024] (D) Parameters :
[15:03:40 - 04/03/2024] (D)    - baseband_format : "f32"
[15:03:40 - 04/03/2024] (D)    - buffer_size : 1000000
[15:03:40 - 04/03/2024] (D)    - dc_block : true
[15:03:40 - 04/03/2024] (D)    - freq_shift : 0
[15:03:40 - 04/03/2024] (D)    - iq_swap : false
[15:03:40 - 04/03/2024] (D)    - samplerate : 10000000
[15:03:40 - 04/03/2024] (D)    - start_timestamp : 1709564620.0
[15:03:40 - 04/03/2024] (I) Module meteor_hrpt_decoder
[15:03:40 - 04/03/2024] (D) Parameters :
[15:03:40 - 04/03/2024] (D)    - baseband_format : "f32"
[15:03:40 - 04/03/2024] (D)    - buffer_size : 1000000
[15:03:40 - 04/03/2024] (D)    - dc_block : true
[15:03:40 - 04/03/2024] (D)    - freq_shift : 0
[15:03:40 - 04/03/2024] (D)    - iq_swap : false
[15:03:40 - 04/03/2024] (D)    - samplerate : 10000000
[15:03:40 - 04/03/2024] (D)    - start_timestamp : 1709564620.0
[15:03:40 - 04/03/2024] (D) Input SPS : 15.028554
[15:03:40 - 04/03/2024] (D) Resample : 1
[15:03:40 - 04/03/2024] (D) Samplerate : 6000000.000000
[15:03:40 - 04/03/2024] (D) Dec factor : 1.666667
[15:03:40 - 04/03/2024] (D) Final SPS : 9.017133
[15:03:40 - 04/03/2024] (I) Start processing...
[15:03:40 - 04/03/2024] (I) Using input baseband 
[15:03:40 - 04/03/2024] (I) Demodulating to /home/orangepi/nas/Discovery_Dish/live_output/2024-03-04_15-03_meteor_hrpt_1.7 GHz/meteor_hrpt.soft
[15:03:40 - 04/03/2024] (I) Buffer size : 163840
[15:03:40 - 04/03/2024] (I) Start processing...
[15:03:40 - 04/03/2024] (I) Progress nan%, SNR : 0.000000dB, Peak SNR: 0.000000dB
[15:03:40 - 04/03/2024] (I) Using input data 
[15:03:40 - 04/03/2024] (I) Decoding to /home/orangepi/nas/Discovery_Dish/live_output/2024-03-04_15-03_meteor_hrpt_1.7 GHz/meteor_hrpt.cadu
[15:03:40 - 04/03/2024] (I) Progress nan%, Deframer : NOSYNC, Frames : 0
[15:03:40 - 04/03/2024] (I) Recording to /home/orangepi/nas/Discovery_Dish/live_output/2024-03-04_15-03-40_10000000SPS_1700000000Hz.f32
[15:03:40 - 04/03/2024] (T) Updating elements...
[... 316 lines cut ...]
[15:19:45 - 04/03/2024] (T) Tried to save empty JPEG!
[15:19:45 - 04/03/2024] (E) None of the required channels are present!
[15:19:45 - 04/03/2024] (I) Saving /home/orangepi/nas/Discovery_Dish/live_output/2024-03-04_15-03_meteor_hrpt_1.7 GHz/MSU-MR/msu_mr_rgb_Natural_Color.jpg...
[15:19:45 - 04/03/2024] (T) Tried to save empty JPEG!
[15:19:45 - 04/03/2024] (E) None of the required channels are present!
[15:19:45 - 04/03/2024] (I) Saving /home/orangepi/nas/Discovery_Dish/live_output/2024-03-04_15-03_meteor_hrpt_1.7 GHz/MSU-MR/msu_mr_rgb_Thermal_Channel.jpg...
[15:19:45 - 04/03/2024] (T) Tried to save empty JPEG!
[15:19:45 - 04/03/2024] (E) None of the required channels are present!
[15:19:45 - 04/03/2024] (I) Saving /home/orangepi/nas/Discovery_Dish/live_output/2024-03-04_15-03_meteor_hrpt_1.7 GHz/MSU-MR/msu_mr_rgb_Day_Microphysics.jpg...
[15:19:45 - 04/03/2024] (T) Tried to save empty JPEG!
[15:19:45 - 04/03/2024] (E) None of the required channels are present!
[15:19:45 - 04/03/2024] (I) Saving /home/orangepi/nas/Discovery_Dish/live_output/2024-03-04_15-03_meteor_hrpt_1.7 GHz/MSU-MR/msu_mr_rgb_Night_Microphysics.jpg...
[15:19:45 - 04/03/2024] (T) Tried to save empty JPEG!
[15:19:45 - 04/03/2024] (E) None of the required channels are present!
[15:19:45 - 04/03/2024] (I) Saving /home/orangepi/nas/Discovery_Dish/live_output/2024-03-04_15-03_meteor_hrpt_1.7 GHz/MSU-MR/msu_mr_rgb_654.jpg...
[15:19:45 - 04/03/2024] (T) Tried to save empty JPEG!
[15:19:45 - 04/03/2024] (E) None of the required channels are present!
[15:19:45 - 04/03/2024] (I) Saving /home/orangepi/nas/Discovery_Dish/live_output/2024-03-04_15-03_meteor_hrpt_1.7 GHz/MSU-MR/msu_mr_rgb_456.jpg...
[15:19:45 - 04/03/2024] (T) Tried to save empty JPEG!
[15:19:45 - 04/03/2024] (E) None of the required channels are present!
[15:19:45 - 04/03/2024] (I) Saving /home/orangepi/nas/Discovery_Dish/live_output/2024-03-04_15-03_meteor_hrpt_1.7 GHz/MSU-MR/msu_mr_rgb_NO_enhancement.jpg...
[15:19:45 - 04/03/2024] (T) Tried to save empty JPEG!
satdump-ui: /home/orangepi/SatDump/src-core/./nlohmann/json.hpp:21357: const value_type& nlohmann::json_abi_v3_11_2::basic_json<ObjectType, ArrayType, StringType, BooleanType, NumberIntegerType, NumberUnsignedType, NumberFloatType, AllocatorType, JSONSerializer, BinaryType, CustomBaseClass>::operator[](const typename nlohmann::json_abi_v3_11_2::basic_json<ObjectType, ArrayType, StringType, BooleanType, NumberIntegerType, NumberUnsignedType, NumberFloatType, AllocatorType, JSONSerializer, BinaryType, CustomBaseClass>::object_t::key_type&) const [with ObjectType = std::map; ArrayType = std::vector; StringType = std::__cxx11::basic_string<char>; BooleanType = bool; NumberIntegerType = long int; NumberUnsignedType = long unsigned int; NumberFloatType = double; AllocatorType = std::allocator; JSONSerializer = nlohmann::json_abi_v3_11_2::adl_serializer; BinaryType = std::vector<unsigned char>; CustomBaseClass = void; nlohmann::json_abi_v3_11_2::basic_json<ObjectType, ArrayType, StringType, BooleanType, NumberIntegerType, NumberUnsignedType, NumberFloatType, AllocatorType, JSONSerializer, BinaryType, CustomBaseClass>::const_reference = const nlohmann::json_abi_v3_11_2::basic_json<>&; typename nlohmann::json_abi_v3_11_2::basic_json<ObjectType, ArrayType, StringType, BooleanType, NumberIntegerType, NumberUnsignedType, NumberFloatType, AllocatorType, JSONSerializer, BinaryType, CustomBaseClass>::object_t::key_type = std::__cxx11::basic_string<char>; nlohmann::json_abi_v3_11_2::basic_json<ObjectType, ArrayType, StringType, BooleanType, NumberIntegerType, NumberUnsignedType, NumberFloatType, AllocatorType, JSONSerializer, BinaryType, CustomBaseClass>::object_t = std::map<std::__cxx11::basic_string<char>, nlohmann::json_abi_v3_11_2::basic_json<>, std::less<void>, std::allocator<std::pair<const std::__cxx11::basic_string<char>, nlohmann::json_abi_v3_11_2::basic_json<> > > >]: Assertion `it != m_value.object->end()' failed.
Aborted (core dumped)
```

JVital2013, 2024-03-23:
Hello again @krakenrf!

I believe I just pushed a fix to this issue - it's tricky to nail down because it doesn't happen every time, but I found a race condition that can lead to this exact crash in the scheduler.

The log makes it look like it's crashing when making composites, but in this situation, composites and updating the next pass happen on different threads, so the order of things can appear mixed up. The crash happens at the same time as "Update pass trajectory...", which is similar to what I'm seeing on my end.

Aang23, 2024-04-23:
@krakenrf did this ever happen again?

krakenrf, 2024-04-27:
Sorry forgot to reply, it appears to be fixed. I haven't had it occur again.

closed 2024-04-27 by krakenrf