SatDump/SatDump #660: Satdump stopped running overnight saying it received a command

JB-Micro, 2024-03-09:
**Description of the issue**
I am running a GOES HRIT setups with Satdump running 24/7 on a Pi4. Late last night, my station stopped uploading data. In my logs, Satdump says it received a command to shut down. When I ran Top in the command line, it showed Satdump was still running but not decoding/saving any data. I was not connected to the Pis with other computers.

**Hardware (SDR/PC/OS)**
RTL-SDR V3, Raspberry Pi4 Raspbian GNU/Linux 11 (bullseye)

**Version (Eg, 1.0.0, CI Build 171)**
Current Master Branch, installed last week.

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)
[satdump-20240308T180516-28.log](https://github.com/SatDump/SatDump/files/14547200/satdump-20240308T180516-28.log)

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**
[image]
The image is the bottom of my log file showing Satdump stopping by itself. I've had Satdump crash before but it never shut down gracefully like this. 

I have my pi reboot every night near midnight using crontab, and have it automatically start Satdump using a python script. So Satdump had only been running a few hours when this occurred.

JVital2013, 2024-03-09:
Hello, thanks for the report!

What does `sudo journalctl --since "2 days ago" | grep -i oom` say? Does it mention that the oom killer (out of memory killer) killed SatDump at the same time? If not, It might be worth looking at the journalctl log itself around the same time, to see if there's anything else in there that might indicate what sent a signal.

JB-Micro, 2024-03-09:
I just dug into the journals and logs and it looks like Satdump was working fine. The time in the logs was throwing me off because I didn't realize there was a 7 hour offset from my system time, but it corresponds to when I set the pi to automatically reboot so that's why it got the signal to stop.

In my journal, I found a USB disconnect error at the time satdump stopped saving images, which corresponds to the Satdump logs for the correct time where nothing was output until I manually closed it this morning.
[image]
[image]

So it seems like no blame is on Satdump for this one, sorry for the report.

closed 2024-03-09 by JB-Micro

JVital2013, 2024-03-09:
No worries, thanks for getting back to us!