SatDump/SatDump #661: Failed to start satdump after Win 11 install

f6bvp, 2024-03-09:
Just started a new mini PC with Windows 11 that was unfortunatly in German that I do not understand.
"Nobody is perfect" !

Download and installed satdump for windows 64 was ok.

Click to start satdump fails with a window in german and I already explained ....

What is wrong ?

[image]

JVital2013, 2024-03-09:
I think we can let it slide that you don't know German ;). It looks like some of your system settings might be set to German now for some reason.

The problem at hand, though, is that it appears your Visual C++ Runtime got removed. You can re-install it from here: https://aka.ms/vs/17/release/vc_redist.x64.exe

f6bvp, 2024-03-09:
Hello,
Thanks for very fast answer and excellent diagnostic. It works ! 
Satdump jumped on my screen.
You are fantastic !
Bernard

closed 2024-03-09 by f6bvp