SatDump/SatDump #662: NOAA APT location calculation error for nightly build v1.2.0-127cab9

X-MQSI, 2024-03-10:
**Description of the issue**
After receiving the NOAA APT, the position of the map overlay was calculated incorrectly.
My ground station is OL50db (China), but after receiving it, the superimposed map is always Antarctica or South America, but the city information identified belongs to Eastern Europe and West Asia...
This error has occurred many times, and in only one of seven receptions, the position was calculated correctly.

**Hardware (SDR/PC/OS)**
SDR：RTL-SDR V4
PC：Desktop computer that I built myself

**Version (Eg, 1.0.0, CI Build 171)**
Nightly Build v1.2.0-127cab9

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)
[00:37:52 - 10/03/2024] (C) LOS!!!!!!!!!!!!!! 25338
[00:37:52 - 10/03/2024] (T) Stop pipeline...
[00:37:52 - 10/03/2024] (I) Stop processing
[00:37:52 - 10/03/2024] (I) Demodulation finished
[00:37:52 - 10/03/2024] (I) Got 1817 lines...
[00:37:52 - 10/03/2024] (I) White balance...
[00:37:53 - 10/03/2024] (T) Update pass trajectory...
[00:37:53 - 10/03/2024] (I) Saving ./live_output/2024-03-10_00-22_noaa_apt_137.62 MHz/raw_unsync.png...
[00:37:53 - 10/03/2024] (I) Synchronize...
[00:37:54 - 10/03/2024] (T) Wedge 1
................
[00:37:54 - 10/03/2024] (C) Couldn't calibrate image with wedges! Likely no valid wedge was identified!
[00:37:54 - 10/03/2024] (I) Saving ./live_output/2024-03-10_00-22_noaa_apt_137.62 MHz/raw_sync.png...
[00:37:54 - 10/03/2024] (I) ----------- Satellite
[00:37:54 - 10/03/2024] (I) NORAD : 25338
[00:37:54 - 10/03/2024] (I) Name  : NOAA-15
[00:37:54 - 10/03/2024] (W) Congratulations for receiving NOAA 15 on APT! It has been 9432 days, 8 hours, 30 minutes and 43 seconds since it has been launched.
[00:37:54 - 10/03/2024] (I) ----------- AVHRR
[00:37:54 - 10/03/2024] (I) Lines : 1817
[00:37:54 - 10/03/2024] (W) Identified channels (debug): A: -1 A1: -1 B: -1
[00:37:54 - 10/03/2024] (I) Saving ./live_output/2024-03-10_00-22_noaa_apt_137.62 MHz/APT-A.png...
[00:37:54 - 10/03/2024] (I) Saving ./live_output/2024-03-10_00-22_noaa_apt_137.62 MHz/APT-B.png...
[00:37:54 - 10/03/2024] (I) Starting processing pipeline noaa_apt...
[00:37:54 - 10/03/2024] (D) Input file (products) : ./live_output/2024-03-10_00-22_noaa_apt_137.62 MHz/dataset.json
[00:37:54 - 10/03/2024] (D) Output file : ./live_output/2024-03-10_00-22_noaa_apt_137.62 MHz
[00:37:54 - 10/03/2024] (D) Starting noaa_apt
[00:37:54 - 10/03/2024] (I) Data is already at level baseband, skipping
[00:37:54 - 10/03/2024] (I) Data is already at level audio_wav, skipping
[00:37:54 - 10/03/2024] (I) Data is already at level products, skipping
[00:37:54 - 10/03/2024] (D) Products processing is enabled! Starting processing module.
[00:37:54 - 10/03/2024] (I) ./live_output/2024-03-10_00-22_noaa_apt_137.62 MHz/.
[00:37:54 - 10/03/2024] (I) ./live_output/2024-03-10_00-22_noaa_apt_137.62 MHz/./product.cbor
[00:37:54 - 10/03/2024] (I) ./live_output/2024-03-10_00-22_noaa_apt_137.62 MHz/./product.cbor
[00:37:54 - 10/03/2024] (I) Loading APT-A.png
[00:37:54 - 10/03/2024] (I) Waiting for the thread...
[00:37:54 - 10/03/2024] (I) Loading APT-B.png
....................
[00:37:54 - 10/03/2024] (D) Composite needs channel cha
[00:37:54 - 10/03/2024] (T) Max Total Width 909
[00:37:54 - 10/03/2024] (T) Max Total Used  909
[00:37:54 - 10/03/2024] (T) Offset for cha is 0
[00:37:55 - 10/03/2024] (I) Saving ./live_output/2024-03-10_00-22_noaa_apt_137.62 MHz/./avhrr_3_APT_channel_A.png...
[00:37:55 - 10/03/2024] (I) Thread stopped
[00:37:55 - 10/03/2024] (I) Saving user config at C:\Users\MQSI\AppData\Roaming/satdump/settings.json
[00:37:55 - 10/03/2024] (D) Composite needs channel chb
[00:37:55 - 10/03/2024] (T) Max Total Width 909
[00:37:55 - 10/03/2024] (T) Max Total Used  909
[00:37:55 - 10/03/2024] (T) Offset for chb is 0
[00:37:55 - 10/03/2024] (I) Saving user config at C:\Users\MQSI\AppData\Roaming/satdump/settings.json
[00:37:55 - 10/03/2024] (T) Updating elements...
[00:37:55 - 10/03/2024] (T) Update pass trajectory...
[00:37:55 - 10/03/2024] (I) Saving ./live_output/2024-03-10_00-22_noaa_apt_137.62 MHz/./avhrr_3_APT_channel_B.png...
[00:37:55 - 10/03/2024] (D) Composite needs channel chb
[00:37:55 - 10/03/2024] (T) Max Total Width 909
[00:37:55 - 10/03/2024] (T) Max Total Used  909
[00:37:55 - 10/03/2024] (T) Offset for chb is 0
[00:37:55 - 10/03/2024] (D) Filtering timestamps...
[00:38:05 - 10/03/2024] (I) Saving ./live_output/2024-03-10_00-22_noaa_apt_137.62 MHz/./avhrr_3_rgb_MCIR_(Uncalibrated).png...
[00:38:05 - 10/03/2024] (D) Filtering timestamps...
[00:38:05 - 10/03/2024] (I) Solving TPS equations...
[00:38:11 - 10/03/2024] (I) Forward solved
[00:38:11 - 10/03/2024] (I) Drawing map overlay...
[00:38:37 - 10/03/2024] (I) Saving ./live_output/2024-03-10_00-22_noaa_apt_137.62 MHz/./avhrr_3_rgb_MCIR_(Uncalibrated)_map.png...
[00:38:37 - 10/03/2024] (D) Composite needs channel cha
[00:38:37 - 10/03/2024] (T) Max Total Width 909
[00:38:37 - 10/03/2024] (T) Max Total Used  909
[00:38:37 - 10/03/2024] (T) Offset for cha is 0
[00:38:37 - 10/03/2024] (D) Filtering timestamps...
[00:38:46 - 10/03/2024] (I) Saving ./live_output/2024-03-10_00-22_noaa_apt_137.62 MHz/./avhrr_3_rgb_MSA_(Uncalibrated).png...
[00:38:47 - 10/03/2024] (I) Applying cached map overlay...
[00:38:47 - 10/03/2024] (I) Saving ./live_output/2024-03-10_00-22_noaa_apt_137.62 MHz/./avhrr_3_rgb_MSA_(Uncalibrated)_map.png...
[00:38:47 - 10/03/2024] (D) Composite needs channel cha
[00:38:47 - 10/03/2024] (D) Composite needs channel chb
[00:38:47 - 10/03/2024] (T) Max Total Width 909
[00:38:47 - 10/03/2024] (T) Max Total Used  909
[00:38:47 - 10/03/2024] (T) Offset for cha is 0
[00:38:47 - 10/03/2024] (T) Offset for chb is 0
[00:38:47 - 10/03/2024] (D) Filtering timestamps...
[00:39:02 - 10/03/2024] (I) Saving ./live_output/2024-03-10_00-22_noaa_apt_137.62 MHz/./avhrr_3_rgb_MCIR_Rain_(Uncalibrated).png...
[00:39:02 - 10/03/2024] (I) Applying cached map overlay...
[00:39:02 - 10/03/2024] (I) Saving ./live_output/2024-03-10_00-22_noaa_apt_137.62 MHz/./avhrr_3_rgb_MCIR_Rain_(Uncalibrated)_map.png...
[00:39:02 - 10/03/2024] (I) Done! Goodbye

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**
[image]
[image]

JVital2013, 2024-03-12:
Hello!

It looks like you are using the automatic scheduler. Have you configured the satellite number within pipeline options? This must be configured for each satellite right now to ensure proper map alignment

[image]

X-MQSI, 2024-03-13:
> Hello!
>
> It looks like you are using the automatic scheduler. Have you configured the satellite number within pipeline options? This must be configured for each satellite right now to ensure proper [...]

Thank you, I see.
I really didn't configure its serial number, which is my fault.
But I think this option may be useless.Maybe you can consider configuring this option automatically?
(This is just my personal opinion.)

Mraanderson, 2025-08-22:
Is this request obsolete now that APT has ceased?