SatDump/SatDump #665: SNR Logs

thomasgolladay, 2024-03-15:
How can I change the frequency of the SNR in the .logs? I would like to log the SNR every like 5 seconds instead of 10 seconds.

Aang23, 2024-04-23:
As it is now, this is not possible. But it does raise a good question. These logs should perhaps be improved...

thomasgolladay, 2024-04-23:
> As it is now, this is not possible. But it does raise a good question. These logs should perhaps be improved...

I was able to follow the code and I found where this is defined. I was able to modify the code to output the SNR and Peak SNR to the log file.

TurboBG, 2024-08-28:
This is exactly what i want to do too. Where in the code did you add your modifications?

mpentler, 2024-11-11:
@thomasgolladay can you come back to this and share your work here? I figure this may be a simple thing to contribute a PR for.

The logical thing to do is to store in a variable from Settings and have a default defined. This seems like a simple beginner issue for someone - maybe me, we'll see. Or Thomas if he wants it of course.

mpentler, 2024-11-19:
I have implemented this as a test in just the pm_demod, but it is easily portable to other demod modules.

[image]

It worked fine with a quick test.