SatDump/SatDump #667: S-pen doesn't work on Samsung tablet

Cpt-Dingus, 2024-03-18:
**Description of the issue**
Attempting to use the S pen in the application doesn't work, the hovering works fine but the actual touch inputs don't register.

**Hardware (SDR/PC/OS)**
Samsung Galaxy Tab S7, Android 13

**Version (Eg, 1.0.0, CI Build 171)**
1.2.0, happens on all versions

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)
N/A

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**
N/A

JVital2013, 2024-03-23:
This sounds like an issue with the ImGui Android backend. Unfortunately, I do not have an S-Pen to test.

I just updated the Android backend itself. It looks like there were a few changes specifically around styluses. Can you update to this actions build (or later) and let us know if it works? https://github.com/SatDump/SatDump/actions/runs/8404138254. If not, we'll look at other options for troubleshooting.

Cpt-Dingus, 2024-03-23:
Sure, give me a sec to install it

Goodbye my dearest user configs!

Cpt-Dingus, 2024-03-23:
Issue is still present on the build you linked.

Aang23, 2024-05-16:
Honestly this will probably have to be reported to ImGui directly, as it's more of a problem with their backend.

Cpt-Dingus, 2024-09-17:
Sorry for the late reply, but it doesn't seem to be an imgui bug; SDR++ works fine.

AlexandreRouma, 2024-09-17:
I had to specifically fix it for SDR++. ImGui ignores pen inputs. It's just a line or two to change.

AlexandreRouma, 2024-09-18:
Pretty sure it was just that yeah

JVital2013, 2024-10-08:
I had some time to test this fix - It makes right-click events on an actual mouse also trigger a left click event at the same time. I'm not a huge fan of this - it sounds like trouble later on. In reality, though, this does not have any impact in SatDump right now, since right-clicking is only used in a single place (double right-click re-centers an image in the viewer). And, due to how this right-click action is implemented, it still works fine.

So, in short:

- I've implemented SDR++'s fix since its been working fine over there. Can you test if it works in the latest nightly @Cpt-Dingus?
- Since it sounds like it still moves the "mouse" around as-is, I'm wondering if the bug can be fixed with a "gentler" patch. I've implemented a different fix in this actions run, but I cannot test it. If you're able, can you also test this build and let me know if it works? https://github.com/SatDump/SatDump/actions/runs/11228029988

If the test patch in the actions run works, we'll go with that patch. Otherwise, we'll stick with SDR++'s method.

Cpt-Dingus, 2024-10-08:
Hello, I'll test once able, thanks.

Cpt-Dingus, 2024-10-12:
Works properly now, cheers!

closed 2024-10-12 by Cpt-Dingus

JVital2013, 2024-10-14:
Great, glad to hear it works @Cpt-Dingus!

Did you try the specific actions build I linked? I'd like to verify if that one works as well

Cpt-Dingus, 2024-10-14:
Hello, give me a second to download it

Cpt-Dingus, 2024-10-14:
Linked patch doesn't work, latest does though.

JVital2013, 2024-10-14:
OK. We'll keep the patch (really, SDR++'s method - thanks @AlexandreRouma!). If the additional click event on right click causes problems later, we'll deal with it then.

For now, it does not, so I'm happy.

Cpt-Dingus, 2024-10-14:
Yeah, might br worth leaving a comment about that there for future reference