SatDump/SatDump #668: r82xx_write: i2c wr failed=-9 reg=1a len=1

f6bvp, 2024-03-19:
**Description of the issue**
Changing satdump frequency while running is ok except if it is connected to remote 
satdump_sdr_server through fast Internet link as shown with 16ms average  ping statistics.

**Hardware (SDR/PC/OS)**
Remote SDR is Nooelec sawbird Generic RTL2832OEM#0

**Version (Eg, 1.0.0, CI Build 171)**
Satdump_sdr_server v1.2.0 and local satdump v1.2.0-395e1ab3

**Logs ** (satdump.log from ~/.config/satdump or %appdata%\satdump)
When changing local frequency while satdump is running remote sdr_server displays error messages : 
  
r82xx_set_freq: failed=-9
r82xx_write: i2c wr failed=-9 reg=1a len=1
r82xx_set_freq: failed=-9
r82xx_write: i2c wr failed=-9 reg=08 len=1
r82xx_set_freq: failed=-9
[09:29:22 - 19/03/2024] (D) Set RTL-SDR frequency to 1545500000
[09:29:22 - 19/03/2024] (D) Frequency sent 1546500000.000000
r82xx_write: i2c wr failed=-9 reg=1b len=1
r82xx_set_freq: failed=-9
r82xx_write: i2c wr failed=-9 reg=17 len=1
r82xx_set_freq: failed=-9
r82xx_write: i2c wr failed=-9 reg=1a len=1
r82xx_set_freq: failed=-9
r82xx_write: i2c wr failed=-9 reg=17 len=1
r82xx_set_freq: failed=-9
r82xx_write: i2c wr failed=-9 reg=17 len=1
r82xx_set_freq: failed=-9

[image]

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**

f6bvp, 2024-03-21:
Same effect just clicking to activate AGC

Disabled direct sampling mode
[13:20:49 - 21/03/2024] (D) Set RTL-SDR frequency to 1544000000
[13:20:49 - 21/03/2024] (D) Set RTL-SDR Bias to 1
[13:20:49 - 21/03/2024] (D) Set RTL-SDR AGC to 0
[13:20:49 - 21/03/2024] (D) Set RTL-SDR Gain to 0
[13:20:49 - 21/03/2024] (D) Set RTL-SDR PPM Correction to 0
[13:20:49 - 21/03/2024] (I) Source started!
[13:20:49 - 21/03/2024] (T) RTL-SDR Buffer size 32256
r82xx_write: i2c wr failed=-9 reg=05 len=1
r82xx_read: i2c wr failed=-9 reg=00 len=1
r82xx_read: i2c wr failed=-9 reg=00 len=1
r82xx_write: i2c wr failed=-9 reg=07 len=1
r82xx_write: i2c wr failed=-9 reg=05 len=1
r82xx_write: i2c wr failed=-9 reg=07 len=1
r82xx_write: i2c wr failed=-9 reg=07 len=1
r82xx_write: i2c wr failed=-9 reg=0c len=1
[13:21:33 - 21/03/2024] (D) Set RTL-SDR AGC to 1
[13:21:33 - 21/03/2024] (D) Set RTL-SDR Gain to 0
[13:21:34 - 21/03/2024] (I) Waiting for the thread...
[13:21:34 - 21/03/2024] (I) Thread stopped
Reattached kernel driver
[13:21:34 - 21/03/2024] (I) Source stopped!
[13:21:34 - 21/03/2024] (D) Sending source settings

JVital2013, 2024-03-23:
Hello! It sounds like the underlying kernel driver is not detaching properly. Try blacklisting the kernel modules by adding the following lines to `/etc/modprobe.d/blacklist.conf`, then reboot the system running SDR Server and try again:

```conf
blacklist dvb_usb_rtl28xxu
blacklist rtl2832
blacklist rtl2830
```

If this does not work, can you try another program on the same machine to see if the dongle works in that program?

AlexandreRouma, 2024-03-23:
If the server is running on a Raspberry Pi this is likely a classic case of an inadequate power supply.

f6bvp, 2024-03-31:
> If the server is running on a Raspberry Pi this is likely a classic case of an inadequate power supply.

Thanks but Satdump server is running on a Tower PC with Intel core i7 CPU à 2.67GHz and 10 pings toward local satdump client is 4.768 ms on average and 5.265 max.

f6bvp, 2024-03-31:
> Hello! It sounds like the underlying kernel driver is not detaching properly. Try blacklisting the kernel modules by adding the following lines to `/etc/modprobe.d/blacklist.conf`, then reboot the system running SDR Server and try again:
> 
> ```
> blacklist dvb_usb_rtl28xxu
> blacklist rtl2832
> blacklist rtl2830
> ```
> 
> If this does not work, can you try another program on the same machine to see if the dongle works in that program?

Alll those sdr are blacklisted including rtl2838.
Locally satdump do not have i2c errors it is only with remote client.
Other application such as CubicSDR also runs fine when on same PC.

closed 2024-07-30 by f6bvp