SatDump/SatDump #669: Stopping geostationary pipelines manually makes the app freeze and get marked as "not responding" by Android

Cpt-Dingus, 2024-03-19:
**Description of the issue**
As the title says, after hitting `Stop` on pipelines, SatDump begins writing the images received from geostationary satellites. While it's writing them, the app freezes and is marked as unresponsive by android. This is not the case when the pipeline isn't manually stopped and writes images while receiving the next batch.

**Hardware (SDR/PC/OS)**
Samsung Galaxy Tab S7, Android 13

**Version (Eg, 1.0.0, CI Build 171)**
1.2.0, happens on all versions

**Logs after the crash (satdump.log from ~/.config/satdump or %appdata%\satdump)**
N/A

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**
Happens with GVAR and S-VISSR broadcasts.

alsldlflglhljlk, 2024-06-30:
Can confirm this is a problem on Windows as well. Especially problematic when there are several unfinished images.

JVital2013, 2024-08-03:
This has been fixed. Thanks @Aang23!

closed 2024-08-03 by JVital2013