SatDump/SatDump #674: Add support for decoding data transmissions from SNOTEL stations.

Alexrules11, 2024-03-27:
**What would you want to be added to satdump?**
Support for receiving SNOTEL meteor scatter transmissions

**Describe the feature you'd like to have**
I realize that SNOTEL stations mainly utilize Meteor Scatter instead of geostationary satellites but support for decoding data transmissions from SNOTEL stations would be handy.

For anyone who is not familiar with SNOTEL stations here is the Wikipedia article on them: https://en.wikipedia.org/wiki/SNOTEL

Thank you!

Aang23, 2024-03-29:
I won't say yes just like that (yet), but I'm fine with looking into it :-)

Do you have some documentation / recordings etc about SNOTEL so I can take a look at what it'd take to support it?

Alexrules11, 2024-03-29:
> I won't say yes just like that (yet), but I'm fine with looking into it :-)
> 
> Do you have some documentation / recordings etc about SNOTEL so I can take a look at what it'd take to support it?
> 
> PS : As a more general rule SatDump officially focuses rather on satellite downlink signals, but since in many cases this may be transpondeurs with not-so-satellite-oriented signals being retransmitted, or things very useful/similar to satellite data (say, radiosondes, ARGOS...), that line is much harder to define. Therefore if there's something which makes sense to use with/similar enough to satellite data I will not oppose it, rather the opposite.

I do not personally have a recording of SNOTEL transmissions however there is an entry in the Signal ID Wiki that does have a recording and some information on SNOTEL transmissions. You can find that entry here: https://www.sigidwiki.com/wiki/SNOTEL_(Snow_Telemetry)

Aang23, 2024-04-02:
Unfortunately this recording being audio it'll be pretty useless as far as decoding goes. It also doesn't seem like any proper protocol specification is available at first glance.

I'll dig a bit deeper later on.