SatDump/SatDump #675: Error: Could not load font file

ivanbistrovic, 2024-03-28:
This probably isn't a problem of SatDump but how I wanted to build and run the program so I am sorry but I don't know where better to ask.
I use another Linux distro in a VM to compile programs, then I copy the build folder with executable binaries to my primary distro where I try to run the program but without installing it, because I don't want to install dependencies to bloat the system.
I manually added all the necessary shared libraries that SatDump requires to work, modify LD_LIBRARY_PATH environment variable and that part is fine.
When I try to start the app it throws this in the standard output:
"[18:21:32 - 28/03/2024] (E) Usage : ./satdump-ui [downlink] [input_level] [input_file] [output_file_or_directory] [additional options as required]
[18:21:32 - 28/03/2024] (E) Extra options (examples. Any parameter used in modules can be used here) :
[18:21:32 - 28/03/2024] (E)  --samplerate [baseband_samplerate] --baseband_format [f32/i16/i8/w8] --dc_block --iq_swap
[18:21:33 - 28/03/2024] (D) Starting with OpenGL 4.6 (Compatibility Profile) Mesa 23.2.1-1ubuntu3.1~22.04.2
[18:21:33 - 28/03/2024] (D) Max texture size: 16384
[18:21:33 - 28/03/2024] (E) Resources fonts/Roboto-Medium.ttf does not exist!
satdump-ui: /home/**user**/Build_Directory/SatDump-1.1.4/src-core/imgui/imgui_draw.cpp:2239: ImFont* ImFontAtlas::AddFontFromFileTTF(const char*, float, const ImFontConfig*, const ImWchar*): Assertion `(0) && "Could not load font file!"' failed.
Aborted (core dumped)"
So the program works but it can't find the font Roboto-Medium.ttf and start the GUI, so I was wondering how to direct the executable to find it, is there anything like an environment variable for fonts?
What led me to further questions is that in the path shown after "_satdump-ui:_", it actually shows a username of my virtual machine (I intentionally made them different) which makes the path invalid on my primary machine. Is there any instruction I need to send to the compiler to build the program with variable path?
I am really lacking knowledge on compiling the programs, hence so many questions but I am really curious about the solution as it can help me learn more about customization and manual deployment of programs.

Cheers

Aang23, 2024-03-29:
To start with the dumb one : 

Did you copy the `pipelines`/`resources` folders in an appropriate place? (Runtime directory or /usr/share/satdump)

ivanbistrovic, 2024-03-30:
> To start with the dumb one :
> 
> Did you copy the `pipelines`/`resources` folders in an appropriate place? (Runtime directory or /usr/share/satdump)

This actually did help. I copied these folders from the master branch to my build folder and error disappeared.
After that it looked for a configuration file in the subfolder "../share/satdump/satdump_cfg.json" but I put it in the root of my build directory and the GUI showed up. This is solved, thanks!

closed 2024-03-31 by Aang23