SatDump/SatDump #676: Satdump won't start up anymore please help me.

Marconist, 2024-03-31:
**Description of the issue**
SatDump, previously working well, no longer wants to start correctly.
Initially, SatDump seems to start up fine with the entrance window. 
[image]

But after loading the TLE, the installation stops and a black entrance window of Satdump v.1.4 appears.
[image]

What have I tried:
Restart PC:  negative, start SatDump
Satdup removed, restart PC, install Satdump:   negative.

The problem started when the following changes were made to the UserInterface, but I doubt this is the cause.
[image]

**Hardware (SDR/PC/OS)**
Operation system: Windows 11

**Version (Eg, 1.0.0, CI Build 171)**

[image]

Satdump v.1.1.4

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)
No Logs.

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**
does not apply

Aang23, 2024-04-01:
Perhaps try deleting the settings.json in `%appdata%\satdump` ?

Marconist, 2024-04-01:
For me the only solution was to deeply uninstall Satdump. 
I had to set everything up again, but that didn't take much time. **Problem solved**

closed 2024-04-03 by Aang23