SatDump/SatDump #677: SatDump v1.1.4 were no response when loading TLEs

minexixi, 2024-04-03:
**Description of the issue**
After the initial installation,the SatDump v1.1.4 were no response when loading TLEs.
**Hardware (SDR/PC/OS)**
SDRPlay RSP1,i3-4170 16GB RAM,Windows 11 professional x64 21H2
**Version (Eg, 1.0.0, CI Build 171)**
SatDump v1.1.4
**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)
[14:05:16 - 03/04/2024] (E) Usage : D:\Ham\SatDump\satdump-ui.exe [downlink] [input_level] [input_file] [output_file_or_directory] [additional options as required]
[14:05:16 - 03/04/2024] (E) Extra options (examples. Any parameter used in modules can be used here) :
[14:05:16 - 03/04/2024] (E)  --samplerate [baseband_samplerate] --baseband_format [f32/i16/i8/w8] --dc_block --iq_swap
[14:05:16 - 03/04/2024] (D) Starting with OpenGL 3.0.0 - Build 20.19.15.5171
[14:05:16 - 03/04/2024] (D) Max texture size: 16384
[14:05:16 - 03/04/2024] (I)    _____       __  ____                      
[14:05:16 - 03/04/2024] (I)   / ___/____ _/ /_/ __ \__  ______ ___  ____ 
[14:05:16 - 03/04/2024] (I)   \__ \/ __ `/ __/ / / / / / / __ `__ \/ __ \
[14:05:16 - 03/04/2024] (I)  ___/ / /_/ / /_/ /_/ / /_/ / / / / / / /_/ /
[14:05:16 - 03/04/2024] (I) /____/\__,_/\__/_____/\__,_/_/ /_/ /_/ .___/ 
[14:05:16 - 03/04/2024] (I)                                     /_/      
[14:05:16 - 03/04/2024] (I) Starting SatDump v1.1.4
[14:05:16 - 03/04/2024] (I) 
[14:05:16 - 03/04/2024] (I) Loading config satdump_cfg.json
[14:05:16 - 03/04/2024] (W) No user configuration found! Keeping defaults.
[14:05:16 - 03/04/2024] (I) Loading plugins from ./plugins
[14:05:16 - 03/04/2024] (T) Loading plugin ./plugins\aaronia_sdr_support.dll...
[14:05:16 - 03/04/2024] (E) Error loading ./plugins\aaronia_sdr_support.dll! Error : ".\plugins\aaronia_sdr_support.dll": 找不到指定的模块。
[14:05:16 - 03/04/2024] (T) Loading plugin ./plugins\aim_support.dll...
[14:05:16 - 03/04/2024] (T) Plugin aim_support loaded!
[14:05:16 - 03/04/2024] (T) Loading plugin ./plugins\airspyhf_sdr_support.dll...
[14:05:16 - 03/04/2024] (T) Plugin airspyhf_sdr_support loaded!
[14:05:16 - 03/04/2024] (T) Loading plugin ./plugins\airspy_sdr_support.dll...
[14:05:16 - 03/04/2024] (T) Plugin airspy_sdr_support loaded!
[14:05:16 - 03/04/2024] (T) Loading plugin ./plugins\analog_support.dll...
[14:05:16 - 03/04/2024] (T) Plugin analog_support loaded!
[14:05:16 - 03/04/2024] (T) Loading plugin ./plugins\bladerf_sdr_support.dll...
[14:05:16 - 03/04/2024] (T) Plugin bladerf_sdr_support loaded!
[14:05:16 - 03/04/2024] (T) Loading plugin ./plugins\bluewalker3_support.dll...
[14:05:16 - 03/04/2024] (T) Plugin bluewalker3_support loaded!
[14:05:16 - 03/04/2024] (T) Loading plugin ./plugins\cluster_support.dll...
[14:05:16 - 03/04/2024] (T) Plugin cluster_support loaded!
[14:05:16 - 03/04/2024] (T) Loading plugin ./plugins\cubesat_support.dll...
[14:05:16 - 03/04/2024] (T) Plugin cubesat_support loaded!
[14:05:16 - 03/04/2024] (T) Loading plugin ./plugins\dmsp_support.dll...
[14:05:16 - 03/04/2024] (T) Plugin dmsp_support loaded!
[14:05:16 - 03/04/2024] (T) Loading plugin ./plugins\dscovr_support.dll...
[14:05:16 - 03/04/2024] (T) Plugin dscovr_support loaded!
[14:05:16 - 03/04/2024] (T) Loading plugin ./plugins\dvb_support.dll...
[14:05:16 - 03/04/2024] (T) Plugin dvb_support loaded!
[14:05:16 - 03/04/2024] (T) Loading plugin ./plugins\elektro_arktika_support.dll...
[14:05:16 - 03/04/2024] (T) Plugin elektro_arktika_support loaded!
[14:05:16 - 03/04/2024] (T) Loading plugin ./plugins\eos_support.dll...
[14:05:16 - 03/04/2024] (T) Plugin eos_support loaded!
[14:05:16 - 03/04/2024] (T) Loading plugin ./plugins\fengyun2_support.dll...
[14:05:16 - 03/04/2024] (T) Plugin fengyun2_support loaded!
[14:05:16 - 03/04/2024] (T) Loading plugin ./plugins\fengyun3_support.dll...
[14:05:16 - 03/04/2024] (T) Plugin fengyun3_support loaded!
[14:05:16 - 03/04/2024] (T) Loading plugin ./plugins\fengyun4_support.dll...
[14:05:16 - 03/04/2024] (T) Plugin fenyun4_support loaded!
[14:05:16 - 03/04/2024] (T) Loading plugin ./plugins\gcom_support.dll...
[14:05:16 - 03/04/2024] (T) Plugin gcom_support loaded!
[14:05:16 - 03/04/2024] (T) Loading plugin ./plugins\geonetcast_support.dll...
[14:05:16 - 03/04/2024] (T) Plugin geonetcast_support loaded!
[14:05:16 - 03/04/2024] (T) Loading plugin ./plugins\gk2a_support.dll...
[14:05:16 - 03/04/2024] (T) Plugin gk2a_support loaded!
[14:05:16 - 03/04/2024] (T) Loading plugin ./plugins\goes_support.dll...
[14:05:16 - 03/04/2024] (T) Plugin goes_support loaded!
[14:05:16 - 03/04/2024] (T) Loading plugin ./plugins\hackrf_sdr_support.dll...
[14:05:16 - 03/04/2024] (T) Plugin hackrf_sdr_support loaded!
[14:05:16 - 03/04/2024] (T) Loading plugin ./plugins\himawari_support.dll...
[14:05:16 - 03/04/2024] (T) Plugin himawari_support loaded!
[14:05:16 - 03/04/2024] (T) Loading plugin ./plugins\hinode_support.dll...
[... 470 lines cut ...]
[14:05:17 - 03/04/2024] (D)  - stereo_hr
[14:05:17 - 03/04/2024] (D)  - stereo_lr
[14:05:17 - 03/04/2024] (D)  - stratosat_tk1_uhf
[14:05:17 - 03/04/2024] (D)  - npp_hrd
[14:05:17 - 03/04/2024] (D)  - swot_tlm
[14:05:17 - 03/04/2024] (D)  - syracuse3b_tlm
[14:05:17 - 03/04/2024] (D)  - terra_db
[14:05:17 - 03/04/2024] (D)  - test_custom
[14:05:17 - 03/04/2024] (D)  - tiahne1_tlm
[14:05:17 - 03/04/2024] (D)  - tianwen1_link
[14:05:17 - 03/04/2024] (D)  - timed_dump
[14:05:17 - 03/04/2024] (D)  - tubin_x_dump
[14:05:17 - 03/04/2024] (D)  - umka_1_dump
[14:05:17 - 03/04/2024] (D)  - viasat3_tlm_12200
[14:05:17 - 03/04/2024] (D)  - viasat3_tlm
[14:05:17 - 03/04/2024] (D)  - youthsat_dump
[14:05:17 - 03/04/2024] (D)  - yunhai_ahrpt
[14:05:17 - 03/04/2024] (I) Found OpenCL Devices (3) :
[14:05:17 - 03/04/2024] (D)  - NVIDIA P106-100
[14:05:17 - 03/04/2024] (D)  - Intel(R) HD Graphics 4400
[14:05:17 - 03/04/2024] (D)  - Intel(R) Core(TM) i3-4170 CPU @ 3.70GHz
[14:05:17 - 03/04/2024] (I) Loading TLEs from ./config/satdump_tles.txt
[14:05:17 - 03/04/2024] (I) 0 TLEs loaded!

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**

minexixi, 2024-04-03:
The latest available version is v1.0.2

Aang23, 2024-04-03:
@minexixi can you try with a nightly build perhaps? I also can't see the TLE error in question here unfortunately, could you share the full log?

minexixi, 2024-04-03:
> @minexixi can you try with a nightly build perhaps? I also can't see the TLE error in question here unfortunately, could you share the full log?

In fact, this is the full log.
Fortunately, I saw the issue # 638. It suggests that I should try to shut down the proxy software that is occupying port 7890, while I happen to be using Clash. When I changed the default 7890 port (http+socks mixed) of the software to 7891, SatDump started normally. This seems to reveal once again the incompatibility of SatDump with the 7890 port proxy software.

YD9CKH, 2024-04-08:
I'm also face the same issue. Im using Satdump nightly build version. Before it running well, no issue. But yesterday, when i run SatDump, it just show screen update TLE 6090... and it stuck. When i check, Satdump not responding.
I'm try reinstall again, but not solving the issue.
Need your help.

YD9CKH, 2024-04-08:
Log after it stuck>

[11:06:03 - 08/04/2024] (E) Usage : E:\HAM\SatDump-Windows_x64_Portable\satdump-ui.exe [downlink] [input_level] [input_file] [output_file_or_directory] [additional options as required]
[11:06:03 - 08/04/2024] (E) Extra options (examples. Any parameter used in modules can be used here) :
[11:06:03 - 08/04/2024] (E)  --samplerate [baseband_samplerate] --baseband_format [f32/i16/i8/w8] --dc_block --iq_swap
[11:06:03 - 08/04/2024] (D) Starting with OpenGL 3.0.0 - Build 31.0.101.2111
[11:06:03 - 08/04/2024] (D) Max texture size: 16384
[11:06:03 - 08/04/2024] (I)    _____       __  ____                      
[11:06:03 - 08/04/2024] (I)   / ___/____ _/ /_/ __ \__  ______ ___  ____ 
[11:06:03 - 08/04/2024] (I)   \__ \/ __ `/ __/ / / / / / / __ `__ \/ __ \
[11:06:03 - 08/04/2024] (I)  ___/ / /_/ / /_/ /_/ / /_/ / / / / / / /_/ /
[11:06:03 - 08/04/2024] (I) /____/\__,_/\__/_____/\__,_/_/ /_/ /_/ .___/ 
[11:06:03 - 08/04/2024] (I)                                     /_/      
[11:06:03 - 08/04/2024] (I) Starting SatDump v1.2.0-42471fa
[11:06:03 - 08/04/2024] (I) 
[11:06:03 - 08/04/2024] (I) Loading config satdump_cfg.json
[11:06:03 - 08/04/2024] (W) No user configuration found! Keeping defaults.
[11:06:03 - 08/04/2024] (I) Loading plugins from ./plugins
[11:06:03 - 08/04/2024] (T) Loading plugin ./plugins\aaronia_sdr_support.dll...
[11:06:03 - 08/04/2024] (E) Error loading ./plugins\aaronia_sdr_support.dll! Error : ".\plugins\aaronia_sdr_support.dll": The specified module could not be found. => D:\a\SatDump\SatDump\src-core\core\plugin.cpp:19
[11:06:03 - 08/04/2024] (T) Loading plugin ./plugins\aim_support.dll...
[11:06:03 - 08/04/2024] (T) Plugin aim_support loaded!
[11:06:03 - 08/04/2024] (T) Loading plugin ./plugins\airspyhf_sdr_support.dll...
[11:06:03 - 08/04/2024] (T) Plugin airspyhf_sdr_support loaded!
[11:06:03 - 08/04/2024] (T) Loading plugin ./plugins\airspy_sdr_support.dll...
[11:06:03 - 08/04/2024] (T) Plugin airspy_sdr_support loaded!
[11:06:03 - 08/04/2024] (T) Loading plugin ./plugins\analog_support.dll...
[11:06:03 - 08/04/2024] (T) Plugin analog_support loaded!
[11:06:03 - 08/04/2024] (T) Loading plugin ./plugins\bladerf_sdr_support.dll...
[11:06:03 - 08/04/2024] (T) Plugin bladerf_sdr_support loaded!
[11:06:03 - 08/04/2024] (T) Loading plugin ./plugins\bluewalker3_support.dll...
[11:06:03 - 08/04/2024] (T) Plugin bluewalker3_support loaded!
[11:06:03 - 08/04/2024] (T) Loading plugin ./plugins\cluster_support.dll...
[11:06:03 - 08/04/2024] (T) Plugin cluster_support loaded!
[11:06:03 - 08/04/2024] (T) Loading plugin ./plugins\cubesat_support.dll...
[11:06:03 - 08/04/2024] (T) Plugin cubesat_support loaded!
[11:06:03 - 08/04/2024] (T) Loading plugin ./plugins\dmsp_support.dll...
[11:06:03 - 08/04/2024] (T) Plugin dmsp_support loaded!
[11:06:03 - 08/04/2024] (T) Loading plugin ./plugins\dscovr_support.dll...
[11:06:03 - 08/04/2024] (T) Plugin dscovr_support loaded!
[11:06:03 - 08/04/2024] (T) Loading plugin ./plugins\dvb_support.dll...
[11:06:03 - 08/04/2024] (T) Plugin dvb_support loaded!
[11:06:03 - 08/04/2024] (T) Loading plugin ./plugins\elektro_arktika_support.dll...
[11:06:03 - 08/04/2024] (T) Plugin elektro_arktika_support loaded!
[11:06:03 - 08/04/2024] (T) Loading plugin ./plugins\eos_support.dll...
[11:06:03 - 08/04/2024] (T) Plugin eos_support loaded!
[11:06:03 - 08/04/2024] (T) Loading plugin ./plugins\fengyun2_support.dll...
[11:06:03 - 08/04/2024] (T) Plugin fengyun2_support loaded!
[11:06:03 - 08/04/2024] (T) Loading plugin ./plugins\fengyun3_support.dll...
[11:06:03 - 08/04/2024] (T) Plugin fengyun3_support loaded!
[11:06:03 - 08/04/2024] (T) Loading plugin ./plugins\fengyun4_support.dll...
[11:06:03 - 08/04/2024] (T) Plugin fenyun4_support loaded!
[11:06:03 - 08/04/2024] (T) Loading plugin ./plugins\gcom_support.dll...
[11:06:03 - 08/04/2024] (T) Plugin gcom_support loaded!
[11:06:03 - 08/04/2024] (T) Loading plugin ./plugins\geonetcast_support.dll...
[11:06:03 - 08/04/2024] (T) Plugin geonetcast_support loaded!
[11:06:03 - 08/04/2024] (T) Loading plugin ./plugins\gk2a_support.dll...
[11:06:03 - 08/04/2024] (T) Plugin gk2a_support loaded!
[11:06:03 - 08/04/2024] (T) Loading plugin ./plugins\goes_support.dll...
[11:06:03 - 08/04/2024] (T) Plugin goes_support loaded!
[11:06:03 - 08/04/2024] (T) Loading plugin ./plugins\hackrf_sdr_support.dll...
[11:06:03 - 08/04/2024] (T) Plugin hackrf_sdr_support loaded!
[11:06:03 - 08/04/2024] (T) Loading plugin ./plugins\himawari_support.dll...
[11:06:03 - 08/04/2024] (T) Plugin himawari_support loaded!
[11:06:03 - 08/04/2024] (T) Loading plugin ./plugins\hinode_support.dll...
[11:06:03 - 08/04/2024] (T) Plugin hinode_support loaded!
[11:06:03 - 08/04/2024] (T) Loading plugin ./plugins\inmarsat_support.dll...
[11:06:03 - 08/04/2024] (T) Plugin inmarsat_support loaded!
[11:06:03 - 08/04/2024] (T) Loading plugin ./plugins\jason3_support.dll...
[11:06:03 - 08/04/2024] (T) Plugin jason3_support loaded!
[... 476 lines cut ...]
[11:06:03 - 08/04/2024] (D)  - slim_tlm_link
[11:06:03 - 08/04/2024] (D)  - smap_s_link
[11:06:03 - 08/04/2024] (D)  - smos_tlm
[11:06:03 - 08/04/2024] (D)  - starship_tlm
[11:06:03 - 08/04/2024] (D)  - stereo_hr
[11:06:03 - 08/04/2024] (D)  - stereo_lr
[11:06:03 - 08/04/2024] (D)  - stratosat_tk1_uhf
[11:06:03 - 08/04/2024] (D)  - npp_hrd
[11:06:03 - 08/04/2024] (D)  - swot_tlm
[11:06:03 - 08/04/2024] (D)  - syracuse3b_tlm
[11:06:03 - 08/04/2024] (D)  - terra_db
[... cut ...]
[11:06:03 - 08/04/2024] (D)  - timed_dump
[11:06:03 - 08/04/2024] (D)  - tubin_x_dump
[11:06:03 - 08/04/2024] (D)  - umka_1_dump
[11:06:03 - 08/04/2024] (D)  - viasat3_tlm_12200
[11:06:03 - 08/04/2024] (D)  - viasat3_tlm
[11:06:03 - 08/04/2024] (D)  - youthsat_dump
[11:06:03 - 08/04/2024] (D)  - yunhai_ahrpt
[11:06:03 - 08/04/2024] (I) Found OpenCL Devices (1) :
[11:06:03 - 08/04/2024] (D)  - Intel(R) HD Graphics 620
[11:06:03 - 08/04/2024] (I) Loading TLEs from ./config/satdump_tles.txt
[11:06:03 - 08/04/2024] (I) 0 TLEs loaded!

JVital2013, 2024-04-09:
Duplicate of #638. We have a fix in the works; please follow that issue for updates

closed 2024-04-09 by JVital2013