SatDump/SatDump #678: SDRPlay RSPdx not supported

super-15, 2024-04-05:
**Description of the issue**

I just got an SDRPlay RSPdx and it works fine with SDRconnect but not with SatDump. It doesn't show up.

**Hardware (SDR/PC/OS)**

SDR - SDRPlay RSPdx
PC - MacBook Air M2
OS - MacOS 14 Sonoma

**Version (Eg, 1.0.0, CI Build 171)**

1.1.4, straight from the MacOS download page

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**

I don't know if I'm just being stupid and I'm not configuring everything correctly, or if the RSPdx is not supported. I have the SDRPlay API installed properly, too.

tonycart, 2024-04-07:
The precompiled app doesn't support SDRPlay. You have to compile the app yourself for SDRPlay devices.

Aang23, 2024-05-02:
@JVital2013 this is fixed right?

JVital2013, 2024-05-02:
It should be, but I did not verify that specific SDR - @super-15 or could you please verify the latest pre-built app works for you?

It requires that the SDRPlay API 3.14 is installed

Aang23, 2024-05-13:
I've had one person confirm this works.

closed 2024-05-13 by Aang23