SatDump/SatDump #679: Add time zone selector in settings for the scheduler

JimmyeJones, 2024-04-06:
**Add time zone selector in settings for the scheduler**
This feature will greatly help me when preparing for satellite passes

Thanks in advance :)

Aang23, 2024-04-10:
What do you mean exactly? Simply show them for your local timezone as to not have to do the conversion manually?

JVital2013, 2024-07-29:
This has been requested multiple times by many different people, so I just implemented it for the scheduler/autotrack UI only. Since this is display-only, I see no harm in it here - but it will show a timezone abbreviation when enabled to avoid confusion.

All other data (anything logged, saved to a file, etc), should not and will not be in local time in my opinion, as it causes ambiguity. 

[image]

closed 2024-07-29 by JVital2013