SatDump/SatDump #685: Building on Linux and running UI fail

2ck, 2024-04-10:
Hello! I've been trying to build SatDump on two Linux machines, and ran into lots of trouble in the process.
I'm using the following tiny Nix flake to have a reproducible build environment:
```
{
    description = ( "SatDump - A generic satellite data processing software" );

    inputs = {
        nixpkgs = { url = "github:NixOS/nixpkgs/release-23.11"; };
    };

    outputs = { self, nixpkgs }:
    let
        system = "x86_64-linux";
        pkgs = nixpkgs.legacyPackages.${system};
        dependencies = [
            pkgs.git pkgs.gnumake pkgs.cmake pkgs.libgcc pkgs.pkgconf
            pkgs.fftwFloat pkgs.libpng pkgs.libtiff pkgs.jemalloc
            pkgs.volk pkgs.nng pkgs.rtl-sdr pkgs.hackrf pkgs.airspy pkgs.airspyhf
            pkgs.glfw pkgs.gnome.zenity
            pkgs.zstd
        ];
    in
    {
        devShells.${system}.default = pkgs.mkShell {
            buildInputs = dependencies;
        };
    };
}
```
As you can see, I just adapted the dependencies from the README.
Interestingly enough, `mbedtls` is listed as a dependency only for MacOS, which brings me to the first issue.

## Part 1 - Building fails

I generate the Makefiles with `cmake -DCMAKE_BUILD_TYPE=Release ..`
Building then fails, among other things, with linker issues:
```
[ 97%] Linking CXX executable ../satdump
/nix/store/bs8irpchp9yrp2azs3arm0b88mrsip6d-binutils-2.40/bin/ld: ../libsatdump_core.so: undefined reference to `mbedtls_ssl_init'
/nix/store/bs8irpchp9yrp2azs3arm0b88mrsip6d-binutils-2.40/bin/ld: ../libsatdump_core.so: undefined reference to `mbedtls_ssl_set_hostname'
/nix/store/bs8irpchp9yrp2azs3arm0b88mrsip6d-binutils-2.40/bin/ld: ../libsatdump_core.so: undefined reference to `mbedtls_ssl_conf_dbg'
[...]
```
This is resolved by adding `pkgs.mbedtls` to the build dependencies, and changing `src-core/CMakeLists.txt` to link `mbedtls` in the appropriate location
```
diff --git a/src-core/CMakeLists.txt b/src-core/CMakeLists.txt
index 8597913c..b4423a0f 100644
--- a/src-core/CMakeLists.txt
+++ b/src-core/CMakeLists.txt
@@ -149,6 +149,8 @@ else()
     find_library(NNG_LIBRARY nng REQUIRED)
     target_link_libraries(satdump_core PUBLIC ${NNG_LIBRARY})
 
+    target_link_libraries(satdump_core PUBLIC mbedtls)
+
     if(BUILD_ZIQ)
         # zstd
         find_library(ZSTD_LIBRARY zstd REQUIRED)
```

The next error is
```
satdump/plugins/sdr_sources/sdrpp_server_support/sdrpp_server/smgui.cpp: In function 'void SmGui::LeftLabel(const char*)':
satdump/plugins/sdr_sources/sdrpp_server_support/sdrpp_server/smgui.cpp:564:39: error: format not a string literal and no format arguments [-Werror=format-security]
  564 |         if (!serverMode) { ImGui::Text(text); ImGui::SameLine(); return; } // LeftLabel?
      |                            ~~~~~~~~~~~^~~~~~

```

The file `src-core-imgui/imgui_demo.cpp` already contains a bandaid fix which I can just copy.
```
#pragma GCC diagnostic ignored "-Wformat-security"                // warning: format string is not a string literal
```

Now finally, the build completes successfully.
I have tested this on the master branch at `b1087fc240c96542c9911b5347fb1860a66829af`, the nightly at `57cb01f253d0bb70327d9a2c26b68ad5e2f66d80` and the 1.1.4 tag.

## Part 2 - Running `satdump-ui` fails

I'm not sure if this should be a separate issue.
Once I've built SatDump with the modifications from above (this time just on `master`), running `./satdump-ui` fails with GLFW errors:
```
[21:14:43 - 10/04/2024] (E) Usage : ./satdump-ui [downlink] [input_level] [input_file] [output_file_or_directory] [additional options as required]
[21:14:43 - 10/04/2024] (E) Extra options (examples. Any parameter used in modules can be used here) :
[21:14:43 - 10/04/2024] (E)  --samplerate [baseband_samplerate] --baseband_format [f32/i16/i8/w8] --dc_block --iq_swap
[21:14:43 - 10/04/2024] (E) Glfw Error 65542: GLX: No GLXFBConfigs returned
[21:14:43 - 10/04/2024] (E) Glfw Error 65545: GLX: Failed to find a suitable GLXFBConfig
[21:14:43 - 10/04/2024] (W) Could not init GLFW Window; falling back to OpenGL 2.1...
[21:14:43 - 10/04/2024] (E) Glfw Error 65542: GLX: No GLXFBConfigs returned
[21:14:43 - 10/04/2024] (E) Glfw Error 65545: GLX: Failed to find a suitable GLXFBConfig
[21:14:45 - 10/04/2024] (C) Could not init GLFW Window! Exiting
```

This seems strange, as `glxinfo | grep -i "glxfb"` returns `896 GLXFBConfigs:`.
My laptop has an AMD Ryzen 7 PRO 5850U, `glxinfo` also reports `OpenGL version string: 4.6 (Compatibility Profile) Mesa 24.0.3`
Running with `LIBGL_ALWAYS_SOFTWARE=1` does not change anything.

Aang23, 2024-05-16:
I will admin I'm no expert with NixOS, and this definitely sound like something isn't being passed right for OpenGL to work. This could be trying to run it with the wrong glfw backend (eg, wayland vs xorg) or such.

You should not need to link mbedtls directly, instead NNG should be built as a shared library (which it is not by default), and itself link to mbedtls if enabled. This is most likely why you are having to link against mbedtls directly. I need to do the same on Android as there everything's static (but this won't be supported on standard Linux builds).

theverygaming, 2024-05-21:
@2ck 
I happened to come across the same problem while packaging SatDump for my NixOS machines (at least the linking part, I didn't have any OpenGL issues - satdump-ui works fine for me) today, and I happened to stumble upon this issue.

It seems to be working fine for me on nixpkgs commit `4a6b83b05df1a8bd7d99095ec4b4d271f2956b64` (nixos-unstable)

Here's my solution:

```nix
{ config, pkgs, ... }:

let
  satdump_version = "1.2.0";
in
{
  environment.systemPackages =
    let
      satdump = with pkgs;
        stdenv.mkDerivation rec {
          pname = "satdump";
          version = satdump_version;

          src = fetchgit {
            url = "https://github.com/SatDump/SatDump.git";
            rev = satdump_version;
            sha256 = "sha256-QGegi5/geL5U3/ecc3hsdW+gp25UE9fOYVLFJUo/N50=";
          };

          nativeBuildInputs = [ cmake pkg-config ];
          buildInputs = [
            # required deps
            fftwFloat
            libpng
            libtiff
            jemalloc
            volk
            (nng.overrideAttrs (old: {
              cmakeFlags = old.cmakeFlags ++ [ "-DBUILD_SHARED_LIBS=ON" ];
            }))
            rtl-sdr-librtlsdr
            hackrf
            airspy
            airspyhf
            glfw
            gnome.zenity
            zstd

            # optional hw support
            libad9361
            libiio
          ];

          cmakeFlags = [
            "-DCMAKE_BUILD_TYPE=Release"
          ];
        };
    in
    [ satdump ];
}
```

NikoKamtsioras, 2024-12-02:
how would i use this?

theverygaming, 2024-12-02:
@NikoKamtsioras What do you mean how would you use this?

Could you please elaborate, do you use NixOS and want to install SatDump?

NikoKamtsioras, 2024-12-02:
yes, i want to install it.

theverygaming, 2024-12-03:
@NikoKamtsioras well you can take the code i have provided, put it into a .nix file, set the `satdump_version` variable to something more recent (see the SatDump GitHub releases) and then include the created .nix file in your NixOS configuration (using `imports` - like the hardware-configuration.nix)

You can see an example of this in [my NixOS configuration](https://github.com/theverygaming/dotfiles/blob/ae0e1ae5dc6d14b5a43479832d1ba7c33be344c7/nixos/pkgs/custom/satdump.nix) although in my case it's all a little more complicated

If you don't use **the Nix package manager** or **NixOS** the above does not apply, in that case please refer to the [SatDump installation instructions](https://github.com/SatDump/SatDump/?tab=readme-ov-file#building--installing)**

JVital2013, 2025-01-04:
Thanks for the work you're putting into this @theverygaming for NixOS. We'll keep an eye on https://github.com/NixOS/nixpkgs/pull/369540.

theverygaming, 2025-05-21:
As the SatDump PR in nixpkgs has been merged now, i believe this issue can be closed!!

SatDump is now available in nixpkgs unstable!