SatDump/SatDump #690: Overlay incorrect.

Galoosis, 2024-04-17:
Although my location is correct the overlay and location for some passes of NOAA are wrong. I am in Washington state and it is showing the passes over Africa and the Antarctic. Images are saved upside down.

RSP1a/Ryzen 6 core/Windows10 pro desktop

Version v1.2.0-c899f09

![channel_4_projected](https://github.com/SatDump/SatDump
[image]
/assets/78892218/d8c2d251-15a1-4561-ab23-a2441a355e2e)
[image]

[image]

Galoosis, 2024-04-21:
I now have it installed on 3 Windows10 pc's. Same issues on all 3. It crashes after a short amount of time requiring task manager to close it.

Aang23, 2024-04-23:
Are the times in the viewer for the passes correct? And are TLEs up-to-date?

JVital2013, 2024-04-23:
Also, make sure you have the right satellite selected here:

[image]

Galoosis, 2024-04-24:
> Are the times in the viewer for the passes correct? And are TLEs up-to-date?

Yes, the times are correct and the TLE's are up to date.

Galoosis, 2024-04-24:
> Also, make sure you have the right satellite selected here: [...]

This I did not have set correctly. Thanks for this info. I did figure out why it was crashing I think. I updated the SDRplay API. It has been running for a little over 24 hours.

JVital2013, 2024-04-27:
Ah yes, the SDRplay API... we've had many headaches with this. Is it still OK after updating, several days later?

Galoosis, 2024-04-27:
Yes, it is still running. The overlay issue is also good now. The NOAA images are saved upside down which is not a big issue. It would be neat if the settings were saved when the program is closed. Some of us would love for it to automatically run using the last settings when opened. Thanks for your help.

JVital2013, 2024-04-29:
Are the upside down ones night passes? If so, this is "normal" - at least for now. This is because the images are saved in the same orientation "as the satellite sees it", and at night POES satellites go south to north. We've had requests before to re-orient the passes automatically, but that has a lot of internal implications to consider.

To make it automatically start autotrack when launching SatDump UI, you should be able to modify the shortcut and add...

```
--start_autotrack --start_recorder_device`
```

...to auto-start the SDR and tracking.

[image]

closed 2024-05-13 by Aang23