SatDump/SatDump #691: Windows CLI Mode

ve3elb, 2024-04-17:
@JVital2013  when running SatDump in CLI mode after a while I get this message and SatDump stops working I have to reboot to get it working again. What can I do to fix this.

[image]

burgeda, 2024-04-18:
Hi Vince

From the looks of that you're running on a raspberry pi.
I had that EXACT same message for months, tried unplugging my rtl-sdr dongle, plugging it back in and the problem would go away for a random amount of time.  Sometimes it'd hang again in minutes other times it would take days but eventually the exact same error reoccurred.  Reinstalled the OS, recompiled the software, tried the distro drivers and also downloaded rtlsdr from github and built that.  No joy.

I even used this attached systemd service file to bump my usbfs memory up to 256M, thought that fixed it but a few days later it reoccurred.

So I broke down and followed what turns out to be common advice and changed my usb cable for  a better one.  Three months later I haven't had the error again.

So, it may sound silly but ... try swapping the usb cable.  Worked WONDERS for me.

Dave - VE3NNW
[usbfs_memory_mb.service.tar.gz](https://github.com/SatDump/SatDump/files/15024834/usbfs_memory_mb.service.tar.gz)

JVital2013, 2024-05-07:
Hi @ve3elb, sorry for the late response. In the past when I've seen this issue, it has indeed been due to a faulty USB connection. Has switching to another USB port, or a new cable, helped your issue?

ve3elb, 2024-05-09:
@JVital2013  Yes I did fix the issue as @burgeda mentioned it was the usb extension cable I was using I replaced it and it's been working fine. Thank you very much.

closed 2024-05-09 by ve3elb