SatDump/SatDump #696: GOES reception in Windows vs Linux

ImDroided, 2024-04-24:
I was hoping someone could help me out here. I have the Nooelec GOES dish, sawbird and a RTL-SDR dongle in windows with SatDump receiving GOES just fine but when I take my setup and plug it into my Raspberry Pi 5 8gb I get no signal lock in SatDump running on Dragon OS. I already updated to the nightly build and I still get no signal lock. I have messed with the gain and had it on auto which works fine in windows. What am I missing here? All my settings are the same between the 2. Any help would be appreciated.

Aang23, 2024-04-24:
Could you show a screenshot perhaps?

ImDroided, 2024-04-24:
Here you go The screenshot in sync is on windows and the other is in Dragon OS running latest nightly on both. All I did is unplug the dongle from the Linux machine and plugged it into my laptop with windows 10. Like I have said I tried messing with the gain and no difference. I dont get why I have a instant sync and decode in windows but nothing in linux. 

[image]
[image]

ImDroided, 2024-04-24:
I jusr realized I didnt have the dongle settings showing

[image]

Aang23, 2024-04-24:
Clearly, there's no signal showing at all. Did you try playing around with the gain on Linux? And see if by any chance the Bias-tee option actually works and similar?

ImDroided, 2024-04-25:
> Clearly, there's no signal showing at all. Did you try playing around with the gain on Linux? And see if by any chance the Bias-tee option actually works and similar?

As stated above I have tried all gain settings and I did it one number at a time. I can see the satellite just fine in SDR++ and CubicSDR so I am not sure what direction to go here. I dont have any issues with NOAA satellites on the same RTL with the nooelec NOAA sawbird. I tried to start SDRPP --server hoping maybe running the rtl remotely per say but still in SatDump I had nothing. I started up sdrpp gui and connected to the sdrpp server and I could see the sat screaming at me.

Aang23, 2024-04-27:
This is really weird. @JVital2013 do you have any idea? Which RTL dongle are you using?

JVital2013, 2024-04-27:
Also, what version of librtlsdr are you using? I've seen other reports of some software having issues with RTL-SDR on the latest version of DragonOS

ImDroided, 2024-04-28:
How do I figure out which version of librtlsdr I'm running?

Aang23, 2024-05-02:
Unfortunately there isn't really an easy option for that. You may be better off uninstalling whatever you have at the moment and building manually, then re-building SatDump. If it turns out to be a problem in DragonOS I'll report it however.

alsldlflglhljlk, 2024-06-30:
@ImDroided Most RTL dongles have a maximum 2.56 MHz usable bandwidth. Try switching to that (on both Windows and DragonOS) and see if you can see the DCS signal on the left

[image]

JimmyeJones, 2024-07-16:
I had the same issue. For me it seemed to be specific to my rtl-sdr… it works fine with my RSP1A from SDRplay. It would be helpful to be able to use an rtl-sdr though…

JVital2013, 2024-07-29:
Following up on this: what RTL-SDR dongle are you both using? I don't believe we ever got an answer on that.

If it's the RTL-SDR Blog v4, then you likely need to re-build the driver from source as not all OSs have updated it yet to be compatible.

JVital2013, 2024-07-29:
Keep in mind the nesder smart v5 does not have a Bias-T, and your amplifier will need to be externally powered with a DC block between the SDR and amplifier

JimmyeJones, 2024-07-29:
I had been powering the amplifier through usb, but the weird thing is it works perfectly fine on my windows laptop…

JVital2013, 2024-07-29:
OK it's not a power issue to the amplifier then. Let me know how rebuilding it works for you. 4 months was a long time ago, we changed a lot. If you have issues, please send your own screenshots and a log

JimmyeJones, 2024-07-29:
I rebuilt satdump like a week ago. I built the OS 4 months ago. Sorry for the confusion. I will rebuild both.

ImDroided, 2024-08-05:
Sorry guys I've been out of town. I just saw this Jamie. I'm using a V4
dongle and yeah it was driver issue related. I don't know if Dragon OS has
older drivers or what, but I was having massive issues with V4 dongles and
the older drivers even though they claimed they were compatible. I did a
fresh pull of the RTL GitHub and compile them from source. Didn't have an
issue after that so something on the Dragon OS build at least whatever
build I was using had an issue. And yeah I had the same thing going on as
the other poster here. Windows dongle perfect lock if I even grabbed a V3
dongle perfect lock V4 dongle gave no signal.  Hopefully alpha Fox will
update his RTL drivers upon next release.

ImDroided, 2024-08-05:
Now I should add here something is going on because if you do a sudo apt
install rtl-sdr The drivers will not work with a V4 you have to compile
from source. I don't know why but that was my issue. I was like man. I'm
installing the drivers apt. Should have the latest stuff or at least close
to latest, but apparently that's not the case. Whatever is going through,
the distribution systems is still broken. I haven't been able to truly
track down the root cause of this yet, but something's weird with the V4
dongles and the drivers.