SatDump/SatDump #697: Tracking module: Meridian flip

sv1, 2024-04-25:
During satellite passes that cross ground station meridian the rotor is repositioning instead of following the satellite.

[image]

sv1, 2024-04-25:
satnogs v2.1

sv1, 2024-04-25:
No I do not. Until now I was using gpredict to control the rotator. From what I can see when there is trans meridian pass, gpredict commands the rotator instead of going to -30d to go to az 150 end el 180.

kng, 2024-05-01:
gpredict does support the "flip" by putting the alt to 180-angle and adding 180 to az, don't remember how it is predicted thou.
I did implement this in two different ways on [satnogs-client](https://gitlab.com/knegge/satnogs-client/-/blob/rotator-flip/satnogsclient/observer/worker.py?ref_type=heads#L135-156) and [ground-station](https://gitlab.com/knegge/ground-station/-/blob/main/python/worker.py?ref_type=heads#L188-189)

Aang23, 2024-05-07:
@kng Ok, that confirms what I thought initially. Many rotators all handle this differently however... I don't have a Satnogs to test so I'll most likely implement it and ask if someone is able to test.