SatDump/SatDump #703: SatDump on Raspberry PI 4B

vu3tyg, 2024-05-01:
Hi,

I am trying to setup SatDump on a Raspberry PI 4B with 4 GB RAM. I have followed the steps as described in https://www.a-centauri.com/articoli/automating-satellite-reception-with-satdump-cli and also installed the dependencies as below

# Install dependencies on Debian-based systems:
sudo apt install git build-essential cmake g++ pkgconf libfftw3-dev libpng-dev libtiff-dev libjemalloc-dev   # Core dependencies
sudo apt install libvolk2-dev                                                                                # If this package is not found, use libvolk-dev or libvolk1-dev
sudo apt install libnng-dev                                                                                  # If this package is not found, follow build instructions below for NNG
sudo apt install librtlsdr-dev libhackrf-dev libairspy-dev libairspyhf-dev                                   # All libraries required for live processing (optional)
sudo apt install libglfw3-dev zenity                                                                         # Only if you want to build the GUI Version (optional)
sudo apt install libzstd-dev                                                                                 # Only if you want to build with ZIQ Recording compression
# (optional)
sudo apt install libomp-dev                                                                                  # Shouldn't be required in general, but in case you have errors with OMP
sudo apt install ocl-icd-opencl-dev                                                                          # Optional, but recommended as it drastically increases speed of some operations. Installs OpenCL.
sudo apt install intel-opencl-icd    

When I try to run SatDump the below errror shows up.  I am not a Linux expert... Any help to fix this will be appreciated

vu3tyg@pi-weather:~ $ satdump autotrack ~/autotrack.json
**_satdump: error while loading shared libraries: libvolk.so.2.4: cannot open shared object file: No such file or directory_**

 Regards
Nitin

dj1471, 2024-05-10:
You installed the build deps, not the runtime deps. For this specific error you need to install libvolk2.5 (assuming Debian/PiOS).

vu3tyg, 2024-05-12:
> You installed the build deps, not the runtime deps. For this specific error you need to install libvolk2.5 (assuming Debian/PiOS).

Thanks for the guidance, Below is what I see after the install

[image]

Aang23, 2024-06-09:
Is this still a problem?

vu3tyg, 2024-06-09:
Hi,
Thanks for checking , this is fixed.
73Nitin [VU3TYG]

Yahoo Mail: Search, organise, conquer

closed 2024-07-28 by JVital2013