SatDump/SatDump #707: Commit - SDR IDs are now strings breaks satdump

burgeda, 2024-05-04:
A commit was made [SDR IDs are now strings] this breaks communication

**Hardware (SDR/PC/OS)**
PI running satdump_sdr_server
Dell micro running satdump

**Version (Eg, 1.0.0, CI Build 171)**

Latest git as of 0930 EDT 2024-05-04

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)

This commandline worked PRIOR to the commit and hasn't worked since

/usr/bin/satdump live goes_hrit /srv/sat-photos --source remote --source_id 0 --samplerate 2.4e6 --frequency 1694.1e6 --gain 10 --agc true --http_server 0.0.0.0:8081

It returns

(E) Error parsing arguments! [json.exception.type_error.302] type must be string, but is number

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**

Running satdump sdr_probe on the pc running satdump I get:

[13:32:58 - 04/05/2024] (I) Querying SDR Server at 192.168.200.230:5656
[13:32:59 - 04/05/2024] (I) Found devices (sources) :
[13:32:59 - 04/05/2024] (I) - File Source
[13:32:59 - 04/05/2024] (D)     - type : file
[13:32:59 - 04/05/2024] (D)     - id : 0
[13:32:59 - 04/05/2024] (I) - Network Source
[13:32:59 - 04/05/2024] (D)     - type : net_source
[13:32:59 - 04/05/2024] (D)     - id : 0
[13:32:59 - 04/05/2024] (I) - 192.168.200.230:5656 - Realtek RTL2838UHIDIR #00000001
[13:32:59 - 04/05/2024] (D)     - type : remote
[13:32:59 - 04/05/2024] (D)     - id : 00000001
[13:32:59 - 04/05/2024] (I) - RTL-TCP
[13:32:59 - 04/05/2024] (D)     - type : rtltcp
[13:32:59 - 04/05/2024] (D)     - id : 0
[13:32:59 - 04/05/2024] (I) - SDR++ Server
[13:32:59 - 04/05/2024] (D)     - type : sdrpp_server
[13:32:59 - 04/05/2024] (D)     - id : 0
[13:32:59 - 04/05/2024] (I) - SpyServer
[13:32:59 - 04/05/2024] (D)     - type : spyserver
[13:32:59 - 04/05/2024] (D)     - id : 0
[13:32:59 - 04/05/2024] (I) Found devices (sinks) :
[13:32:59 - 04/05/2024] (I) Done! Goodbye

Aang23, 2024-05-06:
Can you try the latest, I believe @JVital2013 fixed this.

burgeda, 2024-05-06:
I'll give it a whirl this evening after work and be sure and report.

---- On Mon, 06 May 2024 04:46:26 -0400 Aang23 ***@***.***> wrote ---

Can you try the latest, I believe https://github.com/JVital2013 fixed this.

JVital2013, 2024-05-06:
Sorry, it looks like there's still a bug here @burgeda - I have another SDR with a serial number of 00000001 and it exhibits the problem like you describe. It should be fixed soon though - we'll let you know!

burgeda, 2024-05-06:
I'll give it a try shortly, I'm trying to configure the lastest vitality-goes at the moment.

That example was actually one of several that I tried over the weekend so I was on the right track at least.

Thanks.

---- On Mon, 06 May 2024 16:29:52 -0400 Jamie Vital ***@***.***> wrote ---

OK, this should be fixed for real now https://github.com/burgeda :-). Please try the latest nightly and let us know. Keep in mind you'll need to specify --source_id 00000001 now, since source IDs match the SDR serial number.

burgeda, 2024-05-06:
I've reload both the pi and the pc and it seems to be working just fine now.  I've gotten the updated Vitality-Goes up and displaying the images.

Things are really coming along.

Thanks very much for this absolutely amazing software.

---- On Mon, 06 May 2024 16:29:52 -0400 Jamie Vital ***@***.***> wrote ---

OK, this should be fixed for real now https://github.com/burgeda :-). Please try the latest nightly and let us know. Keep in mind you'll need to specify --source_id 00000001 now, since source IDs match the SDR serial number.

JVital2013, 2024-05-07:
I'm glad to help!

closed 2024-05-07 by JVital2013