SatDump/SatDump #709: No audio to speakers while processing APT

LenStruttmann, 2024-05-05:
**Description of the issue**
While processing APT and with the "Plat Audio" option checked, I still do not get audio out to my speakers.

**Hardware (SDR/PC/OS)**
RTL-SDR V3, Airspy Discovery HF+
SAGER-CLEVO Win11 
Ubunto 22.04

**Version (Eg, 1.0.0, CI Build 171)**
v.1.2.0-26fa5c6 (daily)

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**
Let me know what you'd like to see.

Aang23, 2024-05-06:
Audio will only be generated in "live" processing, not in Offline processing.

LenStruttmann, 2024-05-06:
No, sorry, I am NOT getting any audio during love processing, neither manually nor using Autotrack.

JVital2013, 2024-05-07:
Could you please share your log file? Also, are you using Windows 11, or Ubuntu 22.04? You mention both in your initial post.

LenStruttmann, 2024-05-07:
Sorry for the confusion. I am running Ubuntu 22.04.4 LTS.

[satdump-20240506T181606.log](https://github.com/SatDump/SatDump/files/15235125/satdump-20240506T181606.log)

I hope that this helps.  Thanks!

LenStruttmann, 2024-05-07:
Sorry for the confusion. I am running Ubuntu 22.04.4 LTS
.
[23:16:06 - 06/05/2024] (E) Usage : /usr/bin/satdump-ui [downlink] [input_level] [input_file] [output_file_or_directory] [additional options as required]
[23:16:06 - 06/05/2024] (E) Extra options (examples. Any parameter used in modules can be used here) :
[23:16:06 - 06/05/2024] (E)  --samplerate [baseband_samplerate] --baseband_format [f32/i16/i8/w8] --dc_block --iq_swap
[23:16:06 - 06/05/2024] (D) Starting with OpenGL 4.6 (Compatibility Profile) Mesa 23.2.1-1ubuntu3.1~22.04.2
[23:16:06 - 06/05/2024] (D) Max texture size: 16384
[23:16:06 - 06/05/2024] (I)    _____       __  ____                      
[23:16:06 - 06/05/2024] (I)   / ___/____ _/ /_/ __ \__  ______ ___  ____ 
[23:16:06 - 06/05/2024] (I)   \__ \/ __ `/ __/ / / / / / / __ `__ \/ __ \
[23:16:06 - 06/05/2024] (I)  ___/ / /_/ / /_/ /_/ / /_/ / / / / / / /_/ /
[23:16:06 - 06/05/2024] (I) /____/\__,_/\__/_____/\__,_/_/ /_/ /_/ .___/ 
[23:16:06 - 06/05/2024] (I)                                     /_/      
[23:16:06 - 06/05/2024] (I) Starting SatDump v1.2.0-26fa5c6
[23:16:06 - 06/05/2024] (I) 
[23:16:06 - 06/05/2024] (I) Loading config /usr/share/satdump/satdump_cfg.json
[23:16:06 - 06/05/2024] (I) Loading user settings /home/len/.config/satdump/settings.json
[23:16:06 - 06/05/2024] (I) Loading plugins from /usr/lib/satdump/plugins
[23:16:06 - 06/05/2024] (T) Loading plugin /usr/lib/satdump/plugins/libremote_sdr_support.so...
[23:16:06 - 06/05/2024] (T) Plugin remote_sdr_support loaded!
[23:16:06 - 06/05/2024] (T) Loading plugin /usr/lib/satdump/plugins/librtltcp_support.so...
[23:16:06 - 06/05/2024] (T) Plugin rtltcp_support loaded!
[23:16:06 - 06/05/2024] (T) Loading plugin /usr/lib/satdump/plugins/libproba_support.so...
[23:16:06 - 06/05/2024] (T) Plugin proba_support loaded!
[23:16:06 - 06/05/2024] (T) Loading plugin /usr/lib/satdump/plugins/libcluster_support.so...
[23:16:06 - 06/05/2024] (T) Plugin cluster_support loaded!
[23:16:06 - 06/05/2024] (T) Loading plugin /usr/lib/satdump/plugins/libhinode_support.so...
[23:16:06 - 06/05/2024] (T) Plugin hinode_support loaded!
[23:16:06 - 06/05/2024] (T) Loading plugin /usr/lib/satdump/plugins/libhimawari_support.so...
[23:16:06 - 06/05/2024] (T) Plugin himawari_support loaded!
[23:16:06 - 06/05/2024] (T) Loading plugin /usr/lib/satdump/plugins/libfengyun2_support.so...
[23:16:06 - 06/05/2024] (T) Plugin fengyun2_support loaded!
[23:16:06 - 06/05/2024] (T) Loading plugin /usr/lib/satdump/plugins/libdmsp_support.so...
[23:16:06 - 06/05/2024] (T) Plugin dmsp_support loaded!
[23:16:06 - 06/05/2024] (T) Loading plugin /usr/lib/satdump/plugins/libjpss_support.so...
[23:16:06 - 06/05/2024] (T) Plugin jpss_support loaded!
[23:16:06 - 06/05/2024] (T) Loading plugin /usr/lib/satdump/plugins/libsimd_avx2.so...
[23:16:06 - 06/05/2024] (E) CPU Does not support AVX2. Extension plugin NOT loading!
[23:16:06 - 06/05/2024] (T) Plugin simd_avx2 loaded!
[23:16:06 - 06/05/2024] (T) Loading plugin /usr/lib/satdump/plugins/libsimd_sse41.so...
[23:16:06 - 06/05/2024] (T) Plugin simd_sse41 loaded!
[23:16:06 - 06/05/2024] (T) Loading plugin /usr/lib/satdump/plugins/libcpp_compos_support.so...
[23:16:06 - 06/05/2024] (T) Plugin standard_cpp_compos loaded!
[23:16:06 - 06/05/2024] (T) Loading plugin /usr/lib/satdump/plugins/libgeonetcast_support.so...
[23:16:06 - 06/05/2024] (T) Plugin geonetcast_support loaded!
[23:16:06 - 06/05/2024] (T) Loading plugin /usr/lib/satdump/plugins/liborbcomm_support.so...
[23:16:06 - 06/05/2024] (T) Plugin orbcomm_support loaded!
[23:16:06 - 06/05/2024] (T) Loading plugin /usr/lib/satdump/plugins/libnoaa_metop_support.so...
[23:16:06 - 06/05/2024] (T) Plugin noaa_metop_support loaded!
[23:16:06 - 06/05/2024] (T) Loading plugin /usr/lib/satdump/plugins/libtubsat_support.so...
[23:16:06 - 06/05/2024] (T) Plugin tubsat_support loaded!
[23:16:06 - 06/05/2024] (T) Loading plugin /usr/lib/satdump/plugins/libelektro_arktika_support.so...
[23:16:06 - 06/05/2024] (T) Plugin elektro_arktika_support loaded!
[23:16:06 - 06/05/2024] (T) Loading plugin /usr/lib/satdump/plugins/libinmarsat_support.so...
[23:16:06 - 06/05/2024] (T) Plugin inmarsat_support loaded!
[23:16:06 - 06/05/2024] (T) Loading plugin /usr/lib/satdump/plugins/libdvb_support.so...
[23:16:06 - 06/05/2024] (T) Plugin dvb_support loaded!
[23:16:06 - 06/05/2024] (T) Loading plugin /usr/lib/satdump/plugins/libeos_support.so...
[23:16:06 - 06/05/2024] (T) Plugin eos_support loaded!
[23:16:06 - 06/05/2024] (T) Loading plugin /usr/lib/satdump/plugins/libspyserver_support.so...
[23:16:06 - 06/05/2024] (T) Plugin spyserver_support loaded!
[23:16:06 - 06/05/2024] (T) Loading plugin /usr/lib/satdump/plugins/libgk2a_support.so...
[23:16:06 - 06/05/2024] (T) Plugin gk2a_support loaded!
[23:16:06 - 06/05/2024] (T) Loading plugin /usr/lib/satdump/plugins/librtlsdr_sdr_support.so...
[23:16:06 - 06/05/2024] (T) Plugin rtlsdr_sdr_support loaded!
[23:16:06 - 06/05/2024] (T) Loading plugin /usr/lib/satdump/plugins/libaim_support.so...
[23:16:06 - 06/05/2024] (T) Plugin aim_support loaded!
[23:16:06 - 06/05/2024] (T) Loading plugin /usr/lib/satdump/plugins/liblimesdr_sdr_support.so...
[23:16:06 - 06/05/2024] (T) Plugin limesdr_sdr_support loaded!
[23:16:06 - 06/05/2024] (T) Loading plugin /usr/lib/satdump/plugins/libcubesat_support.so...
[... 817 lines cut ...]
[00:10:40 - 07/05/2024] (D) Skipping 124 as it can't be made!
[00:10:40 - 07/05/2024] (D) Skipping Thermal Channel as it can't be made!
[... cut ...]
[00:10:40 - 07/05/2024] (I) Saving /home/len/Documents/SATdump/live_output/2024-05-06_23-59_noaa_apt_137.62 MHz/./avhrr_3_APT_channel_B.png...
[00:10:41 - 07/05/2024] (I) Saving user config at /home/len/.config/satdump/settings.json
[00:10:41 - 07/05/2024] (I) Saving user config at /home/len/.config/satdump/settings.json
[00:10:41 - 07/05/2024] (D) Composite needs channel chb
[00:10:41 - 07/05/2024] (T) Max Total Width 909
[00:10:41 - 07/05/2024] (T) Max Total Used  909
[00:10:41 - 07/05/2024] (T) Offset for chb is 0
[00:10:41 - 07/05/2024] (I) UI Exit

LenStruttmann, 2024-05-07:
And here is an excerpt from /usr/share/satdump/satdump_cfg.json

        "play_audio": {
            "type": "bool",
            "value": true,
            "name": "Play Audio",
            "description": "Automatically play audio from pipelines that support it,\nlike NOAA APT and Inmarsat Aero"
        },

JVital2013, 2024-05-07:
Whoops, it looks like we have an omission in our build instructions - you need to install `portaudio19-dev` from APT for audio output to work. Install that package, then do a clean build & install.

LenStruttmann, 2024-05-07:
Unfortunately, I am not building from source. I am using a daily builds.  Will there soon be a daily build that fixes this?  Thanks!

f6bvp, 2024-05-08:
After building todays satdump build :
[21:24:39 - 08/05/2024] (T) Server accepted the client
[21:24:39 - 08/05/2024] (T) Waiting for client!
[21:24:39 - 08/05/2024] (E) Error parsing packet from client : basic_string::_M_construct null not valid
[21:24:49 - 08/05/2024] (T) Server lost client
Not sure this is related to portaudio19-dev just installed before compiling satdump-server

JVital2013, 2024-05-08:
@f6bvp this is an unrelated issue. To fix this one, delete everything in /usr/lib/satdump and re-run "make install"

f6bvp, 2024-05-09:
@JVital2013
Thank you. It works.

f6bvp, 2024-05-10:
However clicking on speaker icon turns it into red ad displays "Audio Muted" but does not switch off the sound on PC speaker. 
It just changes the sound pitch.

f6bvp, 2024-05-10:
Same action mutes the sound speaker if device was stoppped and started again and turns is on again with another click.

JVital2013, 2024-07-29:
@LenStruttmann the latest nightly builds now include portaudio support. Let me know if they work for you!

@f6bvp, if you could also let me know if you're still having the issue where the mute button does not work, I'd appreciate it!

LenStruttmann, 2024-08-03:
I installed the nightly build on 03 Aug 2024.  It appears that, YES, I can now hear audio during APT auto-track processing.  

For me, this issue can be closed.  Thanks!  Len

closed 2024-08-03 by JVital2013