SatDump/SatDump #710: Segfault on startup after building in Docker

dj1471, 2024-05-08:
**Description of the issue**
I'm compiling Satdump in Docker on a Pi 4B using the Dockerfile provided. Build works fine without any errors. Upon starting the container image, Satdump segfaults on startup after logging  "Updating elements". If I install the published .deb package into a container image (based on debian:stable-slim) it runs just fine,  I only get the segfault if I self-compile.

**Hardware (SDR/PC/OS)**
Generic RTL2832u, Pi 4B 8GB, PiOS 64-bit

**Version (Eg, 1.0.0, CI Build 171)**
Tested both 1.1.4 and master.

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)
Attached

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**
Starting satdump with: /usr/bin/satdump autotrack /root/autotrack.json
Config and coredump attached
[log.txt](https://github.com/SatDump/SatDump/files/15252220/log.txt)
[core.gz](https://github.com/SatDump/SatDump/files/15252270/core.gz)
[config.txt](https://github.com/SatDump/SatDump/files/15252280/config.txt)

dj1471, 2024-05-28:
Issue still present with 1.2.0. This is a clean build and install into a new container so no issues with bits of older versions being left over per the release notes.

JVital2013, 2024-07-29:
@kng are you able to take a look at this? If not, I can probably set up a test environment

kng, 2024-07-29:
I have not tried the auto tracking in the container, only `satdump-ui` and `satdump_sdr_server`. Tested on arm64 and amd64.
Is there some files it tries to write that fails for some reason.
I know the key to making it work at all with my style of multistage build, is the install prefix and staging prefix.
If it uses the `/target/usr` in anything built it will get into trouble.
```
RUN cmake -B build \
          -DCMAKE_STAGING_PREFIX=/target/usr \
          -DCMAKE_INSTALL_PREFIX=/usr \
```
I'm guessing it is not arch dependant, expecting the same results on amd64.
@JVital2013 I'm not sure how to debug this, so perhaps better you give it a try.

kng, 2024-07-30:
I can at least confirm I got autotrack up and running with the above mentioned json, using the recently merged build package change.
@dj1471 can you test this on your system ?

dj1471, 2024-08-19:
Tested with the current master, no change in behaviour I'm afraid.

Derisis13, 2025-11-13:
I encountered the same problem with lxc containers under proxmox. I have found that just having an RTL-SDR passed trough to the container makes satsump segfault on startup (independent of using it as an input or not). A little investigation with GDB has revealed that the segfaulting code is in libusb, so maybe there's a bad call or a bug in libusb. 

As I've written, the segfault happens even if satdump isn't using the RTL-SDR as its source (I usually use rtl_tcp), so maybe it's doing some kind of enumeration? 
It also doesn't seem to be of concern which user is running satdump: root or unprivileged fail alike. I'be tested satsump built from the master and verywip branches, both segfault the same way.

Other applications (I've tried rtl_sdr, rtl_tcp, ka9q-radio's radiod) seem to have no issue handling the RTL-SDR inside containers, the problem seems to be unique to satdump.

My way of working around this issue was to use a second container to bypass the RTL-SDR to, where I span up rtl_tcp, and used it as a source in my original container. It'll do for the current ARISS experiment, but long term this is hardly a good solution.