SatDump/SatDump #712: Windows version 1.1.4 crash a startup

f6bvp, 2024-05-10:
Immediate crash of satdump v1.1.4 while tentative connexion to remote site.
[image]

After uninstalling and reinstalling I get same crash.
Should I delete some file or directory ? 

Regards,

Bernard

f6bvp, 2024-05-10:
[image]

f6bvp, 2024-05-10:
Crash immediate

JVital2013, 2024-05-12:
Hello again!

Are you mixing and matching versions between client and server? Due to a change the 1.1.4 client is not compatible with the latest 1.2.0 nightly server.

We should probably fix the crash, though. @RAD750 also had this issue

f6bvp, 2024-05-12:
I understand. No problem. I can run Linux too.
Thank you
Bernard

f6bvp, 2024-07-17:
Closed as Windows Satdump 1.2.0 version is starting and doing very nicely.
Thanks a lot !

Bernard

closed 2024-07-17 by f6bvp