SatDump/SatDump #713: Error : Could not find a handler for source type : rtlsdr!

radionugget, 2024-05-18:
Hi, 
I use this [github](https://github.com/jekhokie/raspberry-noaa-v2) in order to receive **NOAA** images with **SatDump.** 
But, I've got this error all the time : 
```
[14:10:01 - 18/05/2024] (E) Could not find a handler for source type : rtlsdr!
```
I don't understand what does it means :( 

This is the line of code with the call to **satdump** : 
```
$SATDUMP live noaa_apt $audio_temporary_storage_directory --source $receiver --samplerate $samplerate --frequency "${NOAA_FREQUENCY}e6" --satellite_number ${SAT_NUMBER} --sdrpp_noise_reduction $gain_option $GAIN $bias_tee_option $crop_topbottom --start_timestamp $PASS_START $finish_processing --timeout $CAPTURE_TIME >> $NOAA_LOG 2>&1
```

Is someone can help me please :) 

---
**Hardware** : 
**RTL-SDR V4** with Raspbian Lite **Bullseye** 64bits

---
**Logs after the crash** :
```
[14:10:01 - 18/05/2024] (I)    _____       __  ____
[14:10:01 - 18/05/2024] (I)   / ___/____ _/ /_/ __ \__  ______ ___  ____
[14:10:01 - 18/05/2024] (I)   \__ \/ __ `/ __/ / / / / / / __ `__ \/ __ \
[14:10:01 - 18/05/2024] (I)  ___/ / /_/ / /_/ /_/ / /_/ / / / / / / /_/ /
[14:10:01 - 18/05/2024] (I) /____/\__,_/\__/_____/\__,_/_/ /_/ /_/ .___/
[14:10:01 - 18/05/2024] (I)                                     /_/
[14:10:01 - 18/05/2024] (I) Starting SatDump v1.2.0-e727f7f3
[14:10:01 - 18/05/2024] (I)
[14:10:01 - 18/05/2024] (I) Loading config /usr/share/satdump/satdump_cfg.json
[14:10:01 - 18/05/2024] (I) Loading user settings /home/nugget/.config/satdump/settings.json
[14:10:01 - 18/05/2024] (I) Loading plugins from /usr/lib/satdump/plugins
...
[14:10:01 - 18/05/2024] (I) Loading TLEs from /home/nugget/.config/satdump/satdump_tles.txt
[14:10:01 - 18/05/2024] (I) 72 TLEs loaded!
[14:10:01 - 18/05/2024] (D) Device File Source
[14:10:01 - 18/05/2024] (D) Device Network Source
[14:10:01 - 18/05/2024] (D) Device RTLSDRBlog Blog V4 #00000001
[14:10:01 - 18/05/2024] (D) Device RTL-TCP
[14:10:01 - 18/05/2024] (D) Device SDR++ Server
[14:10:01 - 18/05/2024] (D) Device SpyServer
[14:10:01 - 18/05/2024] (E) Could not find a handler for source type : rtlsdr!
```

Aang23, 2024-05-18:
Can you check rtl_test works?

radionugget, 2024-05-19:
```
nugget@noaa:~ $ rtl_test
Found 1 device(s):
  0:  RTLSDRBlog, Blog V4, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R828D tuner
RTL-SDR Blog V4 Detected
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6
Sampling at 2048000 S/s.

Info: This tool will continuously read from the device, and report if
samples get lost. If you observe no further output, everything is fine.

Reading samples in async mode...
lost at least 172 bytes
```
I let the command run for 5 minutes

JVital2013, 2024-05-19:
What do you have in the `$receiver` variable? Are you using `--source_id` anywhere?

With the latest update, `--source_id` is now a string that matches the SDR serial number, instead of a device index. If you specify it incorrectly you can get that error.

radionugget, 2024-05-19:
The value of `$receiver` is `rtlsdr`. 
And, `--source_id` is use here : 
```bash
if [[ "$receiver" == "rtlsdr" ]]; then
  gain_option="--source_id $SDR_DEVICE_ID --gain"
```
with `SDR_DEVICE_ID=0`
So, if I understand, I have to replace the **index** `0` with the **SDR serial number** which is `00000001 ` in the screenshot above  ?

JVital2013, 2024-05-19:
That is correct!

closed 2024-05-21 by radionugget