SatDump/SatDump #714: Official v1.2.0 crashes on Win 10

SolarPoweredWok, 2024-05-19:
cAt the every instance of AOS the Satdump just vanishes from the screen i.e closed itself without error or warning.  Even re executing the program and finally ticking "Engage Auto Track",  as soon as AOS at 0 degrees the program vanishes from the screen and is no longer running.
If the satellite is passing over and "Engage Autotrack" is ticked, Satdump again vanished (not running) without notice or message.
Does not matter if in multinode or not.
[satdump-20240520T022825.log](https://github.com/SatDump/SatDump/files/15370298/satdump-20240520T022825.log)
Log file attached but no clue why it closed itself.

 I am running a Desktop PC with a  i7 4GHz CPU 64GB RAM and a Nvidia GeForce RTX 2070 Super GPU with latest Nviia drivers.  No Antivirus or Malware program installed and Windows defender disabled. No other user apps running.
Log file attached.
[satdump-20240520T022825.log](https://github.com/SatDump/SatDump/files/15370298/satdump-20240520T022825.log)
Windows 10 Pro OS 64 bit fully updated.

I have attached the log file from a fresh install of v 1.2.0 Official release, execution to time it closed itself on AOS.

SDR is an SDRPlay RSPdx.  
The API driver comes from the SDRuno Install and is v 3.09 from  SDRUno v1.41.1 as I find later 1.42.xxx versions of SDRUno / API cause Noaa sync to skew severely in Satdump and WxtoImg.

Satdump v 1.2.0 Official release (Not Nightly) 

I loaded a Nightly version (not the latest as it has same problem)  but  v 1.2.0-9c9a525
This nightly version only occasionally crashes when running in live auto mode for Noaa and Meteor Satellites most of the day.

Regards Mike

Aang23, 2024-05-19:
The problem is most likely the config file being incompatible considering some recent changes. Go in %appdata%/satdump and delete settings.json, this should fix it.

SolarPoweredWok, 2024-05-23:
Thanks, deleting the settings.json file and re-entering everything manually again resolved the issue.
I have noted 2 issues with the official release of 1.2.0 
1.  When a filled image is saved, be it Live or Manually processed one cannot use the Viewer to load it to modify the "filled" images with viewer features as not compatible to open.
2.  In the General settings where our Lat/Long location added, one can have a default QTH Label for which I entered my Callsign.  However no Live saves or Manual processes saves will add the QTH label to the images.  It only works when selected with in the viewer functions and I mostly use Live Image saves.  If its in the General setup it should work for all images.
Cheers, Mike VK5LX