SatDump/SatDump #716: Add --v command line option to display Satdump version

f6bvp, 2024-05-21:
**What would you want to be added to satdump?**
Desktop is presently displaying Satdump version on top of main window.

**Describe the feature you'd like to have**
It would be interesting to get this information via command line option --v without actually starting satdump-ui : 
satdump-ui  --v

Grep shows : 
build/satdump.desktop:StartupWMClass=SatDump v1.2.1-36d905a3

Next would be an example of the text to display plus the date of satdump-ui / satdump_sdr_server build like : 
~/satdump$  SatDump v1.2.1-36d905a3 may   21 11:22

Regards,
Bernard

mpentler, 2024-06-15:
If I remember correctly there were command line switches once, because I contributed code around that area, but I think the arguments stuff has changed now.

mpentler, 2024-11-11:
Surely it isn't this simple, in src-core/main.cpp:

    else if (std::string(argv[1]) == "version")
    {
        logger->info("This is SatDump v" + (std::string)SATDUMP_VERSION);
        return 0;
    }

JVital2013, 2024-11-21:
Implemented in #870

closed 2024-11-21 by JVital2013