SatDump/SatDump #717: Only noise on images

radionugget, 2024-05-21:
Hi everyone, I'm unable to get images, only noise... :( 
[image]

I also have this error, is there a link ?  
`(C) Couldn't calibrate image with wedges! Likely no valid wedge was identified`!

This is the line of code with the call to satdump :

```
/usr/bin/satdump live noaa_apt . --source rtlsdr --samplerate 1.024e6 --frequency 137.1000e6 --satellite_number 19 --sdrpp_noise_reduction --source_id 00000001 --gain 29.7 --bias --start_timestamp 1716285483 --finish_processing --timeout 978
```

Is someone can help me please :)

---
Hardware :
RTL-SDR V4 with Raspbian Lite Bullseye 64bits

---
Logs after the crash :

```
[10:12:23 - 21/05/2024] (C) Couldn't calibrate image with wedges! Likely no valid wedge was identified!
[10:12:23 - 21/05/2024] (I) Saving ./raw_sync.png...
[10:12:24 - 21/05/2024] (I) ----------- Satellite
[10:12:24 - 21/05/2024] (I) NORAD : 33591
[10:12:24 - 21/05/2024] (I) Name  : NOAA-19
[10:12:24 - 21/05/2024] (I) ----------- AVHRR
[10:12:24 - 21/05/2024] (I) Lines : 1956
[10:12:24 - 21/05/2024] (W) Identified channels (debug): A: -1 A1: -1 B: -1
[10:12:24 - 21/05/2024] (I) No valid timing data; not aligning timestamp
[10:12:24 - 21/05/2024] (I) Saving ./APT-A.png...
[10:12:24 - 21/05/2024] (I) Saving ./APT-B.png...
[10:12:24 - 21/05/2024] (C) products
[10:12:24 - 21/05/2024] (D) Starting noaa_apt
[10:12:24 - 21/05/2024] (I) Data is already at level baseband, skipping
[10:12:24 - 21/05/2024] (I) Data is already at level audio_wav, skipping
[10:12:24 - 21/05/2024] (I) Data is already at level products, skipping
[10:12:24 - 21/05/2024] (D) Products processing is enabled! Starting processing module.
[10:12:24 - 21/05/2024] (I) ./.
[10:12:24 - 21/05/2024] (I) ././product.cbor
[10:12:24 - 21/05/2024] (I) ././product.cbor
[10:12:24 - 21/05/2024] (I) Loading APT-A.png
[10:12:24 - 21/05/2024] (I) Loading APT-B.png
[10:12:24 - 21/05/2024] (D) Skipping 221 as it can't be made!
[10:12:24 - 21/05/2024] (D) Skipping 3a21 as it can't be made!
[10:12:24 - 21/05/2024] (D) Skipping 4/221 Merge as it can't be made!
[10:12:24 - 21/05/2024] (D) Skipping NOAA Natural Color as it can't be made!
[10:12:24 - 21/05/2024] (D) Skipping Natural Color as it can't be made!
[10:12:24 - 21/05/2024] (D) Skipping 543b as it can't be made!
[10:12:24 - 21/05/2024] (D) Skipping 3b45 as it can't be made!
[10:12:24 - 21/05/2024] (D) Skipping 124 as it can't be made!
[10:12:24 - 21/05/2024] (D) Skipping Thermal Channel as it can't be made!
[10:12:24 - 21/05/2024] (D) Skipping Night Microphysics as it can't be made!
[10:12:24 - 21/05/2024] (D) Skipping Day Microphysics as it can't be made!
[10:12:24 - 21/05/2024] (D) Skipping MCIR as it can't be made!
[10:12:24 - 21/05/2024] (D) Skipping MSA (channel 1) as it can't be made!
[10:12:24 - 21/05/2024] (D) Skipping Sea Surface Temperature as it can't be made!
[10:12:24 - 21/05/2024] (D) Skipping NO enhancement as it can't be made!
[10:12:24 - 21/05/2024] (D) Skipping ZA enhancement as it can't be made!
[10:12:24 - 21/05/2024] (D) Skipping MB enhancement as it can't be made!
[10:12:24 - 21/05/2024] (D) Skipping MD enhancement as it can't be made!
[10:12:24 - 21/05/2024] (D) Skipping BD enhancement as it can't be made!
[10:12:24 - 21/05/2024] (D) Skipping CC enhancement as it can't be made!
[10:12:24 - 21/05/2024] (D) Skipping HE enhancement as it can't be made!
[10:12:24 - 21/05/2024] (D) Skipping HF enhancement as it can't be made!
[10:12:24 - 21/05/2024] (D) Skipping JF enhancement as it can't be made!
[10:12:24 - 21/05/2024] (D) Skipping JJ enhancement as it can't be made!
[10:12:24 - 21/05/2024] (D) Skipping TA enhancement as it can't be made!
[10:12:25 - 21/05/2024] (D) Composite needs channel chb
[10:12:25 - 21/05/2024] (T) Max Total Width 909
[10:12:25 - 21/05/2024] (T) Max Total Used  909
[10:12:25 - 21/05/2024] (T) Offset for chb is 0
[10:12:25 - 21/05/2024] (E) Resources scripted_compos/underlay_with_clouds.lua does not exist!
[10:12:25 - 21/05/2024] (E) Error generating composite! lua: error: cannot open /usr/share/satdump/resources/scripted_compos/underlay_with_clouds.lua: No such file or directory
[10:12:25 - 21/05/2024] (D) Empty image, skipping any further processing!
[10:12:25 - 21/05/2024] (D) Composite needs channel cha
[10:12:25 - 21/05/2024] (T) Max Total Width 909
[10:12:25 - 21/05/2024] (T) Max Total Used  909
[10:12:25 - 21/05/2024] (T) Offset for cha is 0
[10:12:25 - 21/05/2024] (E) Resources scripted_compos/underlay_with_clouds.lua does not exist!
[10:12:25 - 21/05/2024] (E) Error generating composite! lua: error: cannot open /usr/share/satdump/resources/scripted_compos/underlay_with_clouds.lua: No such file or directory
[10:12:25 - 21/05/2024] (D) Empty image, skipping any further processing!
[10:12:25 - 21/05/2024] (D) Composite needs channel cha
[10:12:25 - 21/05/2024] (D) Composite needs channel chb
[10:12:25 - 21/05/2024] (T) Max Total Width 909
[10:12:25 - 21/05/2024] (T) Max Total Used  909
[10:12:25 - 21/05/2024] (T) Offset for cha is 0
[10:12:25 - 21/05/2024] (T) Offset for chb is 0
[10:12:25 - 21/05/2024] (D) Filtering timestamps...
libpng error: Read Error
[10:12:29 - 21/05/2024] (E) Error generating composite! lua: error: ...p/resources/scripted_compos/underlay_with_clouds_lut.lua:49: attempt to perform arithmetic on a nil value (global 'cfg_offset')
[10:12:29 - 21/05/2024] (I) Saving ././avhrr_3_rgb_MCIR_Rain_(Uncalibrated).png...
[10:12:29 - 21/05/2024] (D) Reprojecting composite avhrr_3_rgb_MCIR_Rain_(Uncalibrated)
[10:12:29 - 21/05/2024] (D) Filtering timestamps...
[10:12:30 - 21/05/2024] (T) Warping size 9090x4545
[10:12:30 - 21/05/2024] (T) Smart Warp using 1919 1540, full size 9090 4545
[10:12:30 - 21/05/2024] (T) We will split into 2 segments. Median distance is 95.334500 km and total (avg) distance is 6292.076998 km
[10:12:30 - 21/05/2024] (I) Solving TPS equations for 1089 GCPs...
[10:12:30 - 21/05/2024] (I) Solving TPS equations for 1152 GCPs...
[10:12:36 - 21/05/2024] (I) Solved! Took 6.238465
[10:12:38 - 21/05/2024] (I) Solved! Took 7.835962
[10:12:38 - 21/05/2024] (D) Using CPU!
[10:12:52 - 21/05/2024] (D) CPU Processing Time 13.388670
[10:12:52 - 21/05/2024] (D) Using CPU!
[10:12:59 - 21/05/2024] (D) CPU Processing Time 7.524138
[10:13:06 - 21/05/2024] (I) Saving ././rgb_avhrr_3_rgb_MCIR_Rain_(Uncalibrated)_projected.png...
[10:13:07 - 21/05/2024] (I) Done! Goodbye
```

pkreuzt, 2024-05-28:
I have the same issue. If this helps, I am able to manually set the frequency and pick up the signal just fine. So something in the pipeline must be broken OR the scheduler doesn't set the frequency correctly.

pkreuzt, 2024-05-28:
Also in live mode the image is being decoded. Or at least the miniature shows an image. Watching a live pass from scheduler it doesn't show decoded image on miniature nor peaks on FFT. So my bet is scheduler doesn´t set the frequency correctly.

Aang23, 2024-06-09:
Could you show some screenshots of your configuration and during reception? This is most likely not a SatDump bug at all but some configuration problem or similar.

pkreuzt, 2024-07-01:
I have been experimenting a bit since V4 is new for me. It seems to require maxing gain to receive something. A bias-tee powered LNA also helps. Without maxing gain all the way to the upper limit, all you can get is noise.

Also discovered I had misunderstood the initial frequency in multi mode of autotrack. If multi mode is disabled, you can leave initial frequency at 0, but with multi enabled it is to be set at the center frequency of all intended passes (and with enough bandwith to catch them all).

JVital2013, 2024-07-29:
Hello @nxugget and @pkreuzt! Is this still happening for either of you? If you have not already, please try the latest nightly.

If you are still having issues, could you please post the entire log of a reception where this happens? Thanks!

pkreuzt, 2024-07-29:
@JVital2013 My problem was what I explained on my previous comment here: I had missunderstood the "initial frequency" parameter when in multi mode. Fixing that config it works fine for me.

JVital2013, 2024-07-30:
Gotcha, thanks. Let's see if nxugget has any luch

JVital2013, 2024-09-29:
@nxugget would you like any further assistance with this issue?

nxugget, 2024-09-30:
Hi, sorry, no, I completely restart my entire installation and I don't have the issue anymore :)

closed 2024-09-30 by nxugget